luar123/zigbee_esphome #3: Using esp32-c6-devkitc-1, 8flash development board, log error reported

jjcszxh, 2024-06-13:
I have already used Zigbee's partition table

[2024-06-12 22:16:09.623 R][0;32mI (675) main_task: Returned from app_main()[0m
[0;32m[I][logger:156]: Log initialized[0m

[2024-06-12 22:16:10.167 R]
assert failed: block_trim_free heap_tlsf.c:406 (block_is_free(block) && "block must be free")
Core  0 register dump:
MEPC    : 0x40801d96  RA      : 0x408077d0  SP      : 0x4081bf10  GP      : 0x4080e090  
TP      : 0x40809e98  T0      : 0x73695f6b  T1      : 0x6f6c6222  T2      : 0x636f6c62  
S0/FP   : 0x00000001  S1      : 0x4081c08d  A0      : 0x4081bf68  A1      : 0x4080e71d  
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x40814000  
A6      : 0x00000008  A7      : 0x6d206b63  S2      : 0x0000006a  S3      : 0x4081bf68  
S4      : 0x4081bf68  S5      : 0x40814000  S6      : 0x00000000  S7      : 0x40813e50  
S8      : 0x4081d190  S9      : 0x4208e000  S10     : 0x40814000  S11     : 0x40814000  
T3      : 0x20262620  T4      : 0x296b636f  T5      : 0x6c622865  T6      : 0x6572665f  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
MHARTID : 0x00000000  

Stack memory:
4081bf10: 0xa5a5a5a5 0xa5a5a5a5 0x4004d058 0x4080c97a 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x4080e71c
4081bf30: 0x00363034 0xa5a5a5a5 0x00000000 0x00000000 0xa5000000 0x4080e910 0x4004d058 0x4080e920
4081bf50: 0x4004d150 0x4080e924 0x4081bf30 0x4080e928 0x4004d3b8 0x4080e71c 0x65737361 0x66207472
4081bf70: 0x656c6961 0x62203a64 0x6b636f6c 0x6972745f 0x72665f6d 0x68206565 0x5f706165 0x66736c74
4081bf90: 0x343a632e 0x28203630 0x636f6c62 0x73695f6b 0x6572665f 0x6c622865 0x296b636f 0x20262620
4081bfb0: 0x6f6c6222 0x6d206b63 0x20747375 0x66206562 0x22656572 0x42010029 0x4081c040 0x4080e090
4081bfd0: 0x40809e98 0x5b6d3233 0x20676f4c 0x3b305b1b 0x4080da4c 0x4080f5dc 0x00000001 0x00000a00
4081bff0: 0x0000000a 0xffff00ff 0x00000000 0x20001090 0x4081cbfc 0x00000000 0x4081ce78 0x40814000
4081c010: 0x40814000 0x4208e000 0x4081d190 0x42082000 0x42081000 0x4208126c 0x4081c188 0x4081c1b8
4081c030: 0x4081d49d 0x4081c2cc 0x00000003 0x0000000c 0x00000000 0x40813e64 0x4081d4a8 0x40028392
4081c050: 0x00000000 0x40813e64 0x4081d4a8 0x400213e2 0x00000000 0x40814000 0x00000003 0x00000000
4081c070: 0x40814530 0x40880000 0x40813e50 0x40022542 0x4081d0e4 0x40880000 0x40813e50 0x00000009
4081c090: 0x40814530 0x00000009 0x00001800 0x40802456 0x4081d16c 0x40880000 0x40813e50 0x42082000
4081c0b0: 0x42081000 0x4081d49c 0x4081c188 0x4081c1b8 0x00001800 0x00000009 0x00000009 0x40802482
4081c0d0: 0x4081c1a0 0x4081d48c 0x00000009 0x4206b7d4 0x00000000 0x20001090 0x4081c118 0x420072ca
4081c0f0: 0x40814530 0x4081d48c 0x4081d190 0x42007a34 0x42081000 0x00000000 0x40814000 0x00000000
4081c110: 0x00001800 0x00000008 0x4081c120 0x00000008 0x34323032 0x32313630 0x67697a00 0x00656562
4081c130: 0x42081000 0x4208126c 0x40814000 0x42086000 0x4080f000 0x40814000 0x4080f4f8 0x4200c956
4081c150: 0x4081c1d0 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4081c170: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x4081c190 0x0000000f
4081c190: 0x33707365 0x36632d32 0x67697a2d 0x00656562 0x4081c1a8 0x00000007 0x68707365 0x00656d6f
4081c1b0: 0x67697a2d 0x00656562 0x4081c1c0 0x00000008 0x34323032 0x32313630 0x00000000 0x4080e090
4081c1d0: 0x4081c1d8 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081c1f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081c210: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x4200082c
4081c230: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081c250: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4081c270: 0x00000150 0x4081bfc0 0x00000000 0x4080f724 0x4080f724 0x4081c274 0x4080f71c 0x00000018
4081c290: 0x00000000 0x00000000 0x4081c274 0x00000000 0x00000001 0x4081a270 0x706f6f6c 0x6b736154
4081c2b0: 0x00000000 0x00000000 0x4081c260 0x00000001 0x00000000 0x00000000 0x00000000 0x00000000
4081c2d0: 0x40814930 0x40814998 0x40814a00 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000
4081c2f0: 0x00000000 0x42019b9e 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

ELF file SHA256: 3e09909e2

Rebooting...
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0xc (SW_CPU),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001975a
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xe58
load:0x4086e610,len:0x2e20
load:0x40875720,len:0x1804
entry 0x4086c410
[0;32mI (26) boot: ESP-IDF 5.2.1 2nd stage bootloader[0m
[0;32mI (27) boot: compile time Jun 12 2024 13:52:45[0m
[0;32mI (28) boot: chip revision: v0.0[0m
[0;32mI (30) boot.esp32c6: SPI Speed      : 80MHz[0m
[0;32mI (34) boot.esp32c6: SPI Mode       : DIO[0m
[0;32mI (39) boot.esp32c6: SPI Flash Size : 4MB[0m
[0;32mI (44) boot: Enabling RNG early entropy source...[0m
[0;32mI (49) boot: Partition Table:[0m
[0;32mI (53) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (60) boot:  0 otadata          OTA data         01 00 00009000 00002000[0m
[0;32mI (68) boot:  1 phy_init         RF data          01 01 0000b000 00001000[0m
[0;32mI (75) boot:  2 app0             OTA app          00 10 00010000 001b0000[0m
[0;32mI (82) boot:  3 app1             OTA app          00 11 001c0000 001b0000[0m
[0;32mI (90) boot:  4 nvs              WiFi data        01 02 00370000 0006d000[0m
[0;32mI (97) boot:  5 zb_storage       Unknown data     01 81 003dd000 00004000[0m
[0;32mI (105) boot:  6 zb_fct           Unknown data     01 81 003e1000 00000400[0m
[0;32mI (113) boot: End of partition table[0m
[0;32mI (117) esp_image: segment 0: paddr=00010020 vaddr=42080020 size=13088h ( 77960) map[0m

[2024-06-12 22:16:10.230 R][0;32mI (157) esp_image: segment 1: paddr=000230b0 vaddr=40800000 size=0cf68h ( 53096) load[0m
[0;32mI (183) esp_image: segment 2: paddr=00030020 vaddr=42000020 size=78d0ch (494860) map[0m

[2024-06-12 22:16:10.618 R][0;32mI (386) esp_image: segment 3: paddr=000a8d34 vaddr=4080cf68 size=0091ch (  2332) load[0m
[0;32mI (389) esp_image: segment 4: paddr=000a9658 vaddr=4080d890 size=01c50h (  7248) load[0m
[0;32mI (402) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (403) boot: Disabling RNG early entropy source...[0m
[0;32mI (415) cpu_start: Unicore app[0m
[0;33mW (424) clk: esp_perip_clk_init() has not been implemented yet[0m
[0;32mI (431) cpu_start: Pro cpu start user code[0m
[0;32mI (431) cpu_start: cpu freq: 160000000 Hz[0m
[0;32mI (431) cpu_start: Application information:[0m
[0;32mI (434) cpu_start: Project name:     esp32-c6-zigbee[0m
[0;32mI (439) cpu_start: App version:      2024.5.5[0m
[0;32mI (444) cpu_start: Compile time:     Jun 12 2024 13:52:12[0m
[0;32mI (451) cpu_start: ELF file SHA256:  3e09909e2...[0m
[0;32mI (456) cpu_start: ESP-IDF:          5.2.1[0m
[0;32mI (461) cpu_start: Min chip rev:     v0.0[0m
[0;32mI (465) cpu_start: Max chip rev:     v0.99 [0m
[0;32mI (470) cpu_start: Chip rev:         v0.0[0m
[0;32mI (475) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (482) heap_init: At 40813E50 len 000687C0 (417 KiB): RAM[0m
[0;32mI (488) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM[0m
[0;32mI (494) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM[0m
[0;32mI (501) spi_flash: detected chip: generic[0m
[0;32mI (505) spi_flash: flash io: dio[0m
[0;33mW (509) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.[0m
[0;33mW (522) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`[0m
[0;33mW (533) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`[0m
[0;32mI (544) sleep: Configure to isolate all GPIO pins in sleep state[0m
[0;32mI (551) sleep: Enable automatic switching of GPIO sleep configuration[0m
[0;32mI (558) coexist: coex firmware version: 77cd7f8[0m
[0;32mI (563) coexist: coexist rom version 5b8dcfa[0m
[0;32mI (568) main_task: Started on CPU0[0m
[0;32mI (571) main_task: Calling app_main()[0m

jjcszxh, 2024-06-13:
This may be a memory leak issue{assert failed: block_trim_free heap_tlsf.c:406 (block_is_free(block) && "block must be free")}

luar123, 2024-06-13:
Have you tried to erase flash with `esptool erase_flash`?

JochenFriedrich, 2024-07-07:
I got the same problem, but only on one yaml, a second one works OK. The only difference is the name. Recompiling with a shorter name or without a friedlyname seems to make it work in some cases. Very strange.

commit 66f0ac72 (2024-08-24): Include PRs from @JochenFriedrich and @abmantis:
Fix issue with character strings. Fix #3
Add reporting option to attribute setup.
closed 2024-08-24 by luar123