luar123/zigbee_esphome #4: How to add a light intensity sensor? I followed your writing style, but it doesn't take effect.

lxz946786639, 2024-06-15:
`zigbee:
  id: "zb"
  endpoints:
    - num: 1
      device_type: ON_OFF_LIGHT
      clusters:
        - id: ON_OFF
          attributes:
            - id: 0
              type: bool
              on_value:
                then:
                  - light.control:
                      id: light_1
                      state: !lambda "return (bool)x;"
                      ...
    - device_type: TEMPERATURE_SENSOR
      num: 2
      clusters:
        - id: REL_HUMIDITY_MEASUREMENT
          attributes:
            - id: 0
              type: U16
              value: 200
        - id: TEMP_MEASUREMENT
          attributes:
            - id: 0x0
              type: S16
              value: 100
    - device_type: SIMPLE_SENSOR
      num: 3
      clusters:
        - id: ILLUMINANCE_MEASUREMENT
          attributes:
            - id: 0x1
              type: U16
              value: 0`

luar123, 2024-06-15:
Please use code blocks https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks

I guess the measured value attribute has the ID 0, not 1. Check ZCL, but 1 should be min measured value.

lxz946786639, 2024-06-15:
> Please use code blocks https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks
> 
> I guess the measured value attribute has the ID 0, not 1. Check ZCL, but 1 should be min measured value.

ok， Sorry. I used code blocks.
I made the modifications according to your guess, but the same ones didn't work.
I guess it's an issue with `device_type: SIMPLE-SENSOR`.

···
  id: "zb"
  endpoints:
    - num: 1
      device_type: ON_OFF_LIGHT
      clusters:
        - id: ON_OFF
          attributes:
            - id: 0
              type: bool
              on_value:
                then:
                  - light.control:
                      id: light_1
                      state: !lambda "return (bool)x;"
                      ...
    - device_type: TEMPERATURE_SENSOR
      num: 2
      clusters:
        - id: REL_HUMIDITY_MEASUREMENT
          attributes:
            - id: 0
              type: U16
              value: 200
        - id: TEMP_MEASUREMENT
          attributes:
            - id: 0x0
              type: S16
              value: 100
    - device_type: SIMPLE_SENSOR
      num: 3
      clusters:
        - id: ILLUMINANCE_MEASUREMENT
          attributes:
            - id: 0
              type: U16
              value: 100
···

luar123, 2024-06-15:
Yes Simple sensor is the wrong type, but not sure if this could be an issue. What coordinator, software are you using?

I would just add the illuminance cluster to the temp sensor, because there is no device type for that kind of sensor. Or you could try HA_TEST_DEVICE.

BTW, what do you mean by it is not working? Any logs?

lxz946786639, 2024-06-15:
I am using homeassistant zibgee2mqtt.

Yes, adding it to the temperature sensor is possible.

Tested, `HA_TEST_DEVICE `does not work. In the esphome_zb_default_clusters_create() method in zibgee_helpers. c, devices in the `case `function normally, while `default `does not.

[image]

lxz946786639, 2024-06-15:
> Yes Simple sensor is the wrong type, but not sure if this could be an issue. What coordinator, software are you using?
>
> I would just add the illuminance cluster to the temp sensor, because there is [...]

The value of attributes ID seems to be able to take any value. Where can I find its specific description?

lxz946786639, 2024-06-15:
I add the illuminance cluster to the temp sensor,is work ok.

But I found an issue where the delay in sensor data reporting is very high, and it can take from one minute to a few minutes to wait. It can be reported normally in your esphome_zb_sensor project.

luar123, 2024-06-23:
I will have to see if there is a bug with simple sensor or HA test device. Just no time at the moment...

Which version of z2m are you using?
I was using 1.36 before and that gave me a similar error due to the two endpoints. Automatic device recognition was only working for one endpoint. Updating to 1.38 fixed it, but I never tried with 3 endpoints. In general it is always a good idea to try with one endpoint to simplify the system.

> The value of attributes ID seems to be able to take any value. Where can I find its specific description?

Check the official zigbee cluster lib: https://www.zigbeealliance.org/wp-content/uploads/2021/10/07-5123-07-ZigbeeClusterLibrary_Revision_7-1.pdf

For reporting see: #5

lxz946786639, 2024-06-25:
Ok, thank you.My z2m version is 1.38 . I will try testing it with one endpoint today.

dronix27, 2024-07-19:
I would like to see an example of adding a binary sensor

dronix27, 2024-07-19:
> > Please use code blocks https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks
> > I guess the measured value attribute has the ID 0, not 1. Check ZCL, but 1 should be min measured value.
> 
> ok， Sorry. I used code blocks. I made the modifications according to your guess, but the same ones didn't work. I guess it's an issue with `device_type: SIMPLE-SENSOR`.
> 
> ```
> zigbee:
>   id: "zb"
>   endpoints:
>     - num: 1
>       device_type: COLOR_DIMMABLE_LIGHT
>       clusters:
>         - id: ON_OFF
>           attributes:
>             - id: 0
>               type: bool
>               on_value:
>                 then:
>                   - light.control:
>                       id: light_1
>                       state: !lambda "return (bool)x;"
>         - id: LEVEL_CONTROL
>           attributes:
>             - id: 0
>               type: U8
>               on_value:
>                 then:
>                   - light.control:
>                       id: light_1
>                       brightness: !lambda "return ((float)x)/255;"
>         - id: COLOR_CONTROL
>           attributes:
>             - id: 3
>               type: U16
>               on_value:
>                 then:
>                   - lambda: id(color_x) = (float)x/65536;
>                   - light.control:
>                       id: light_1
>                       red: !lambda "return zigbee::get_r_from_xy(id(color_x), id(color_y));"
>                       green: !lambda "return zigbee::get_g_from_xy(id(color_x), id(color_y));"
>                       blue: !lambda "return zigbee::get_b_from_xy(id(color_x), id(color_y));"
>             - id: 4
>               type: U16
>               on_value:
>                 then:
>                   - lambda: id(color_y) = (float)x/65536;
>                   - light.control:
>                       id: light_1
>                       red: !lambda "return zigbee::get_r_from_xy(id(color_x), id(color_y));"
>                       green: !lambda "return zigbee::get_g_from_xy(id(color_x), id(color_y));"
>                       blue: !lambda "return zigbee::get_b_from_xy(id(color_x), id(color_y));"
>     - device_type: TEMPERATURE_SENSOR
>       num: 2
>       clusters:
>         - id: REL_HUMIDITY_MEASUREMENT
>           attributes:
>             - id: 0
>               type: U16
>               value: 200
>         - id: TEMP_MEASUREMENT
>           attributes:
>             - id: 0x0
>               type: S16
>               value: 100
>     - device_type: SIMPLE_SENSOR
>       num: 3
>       clusters:
>         - id: ILLUMINANCE_MEASUREMENT
>           attributes:
>             - id: 0
>               type: U16
>               value: 1
>   on_join:
>     then:
>       - logger.log: "Joined network"
> ```

Can you show the sensor section?

dronix27, 2024-08-04:
I suggest the following code. Any edits and hints are welcome.

```
sensor:
  - platform: tsl2561
    name: "TSL2561 Ambient Light"
    address: 0x39
    id: "lum"
    on_value:
      then:
        - zigbee.setAttr:
            id: zb
            endpoint: 2
            cluster: ILLUMINANCE_MEASUREMENT
            attribute: 0x0
            value: !lambda "return 10000*log10 (x+1);"

zigbee:
  id: "zb"
  endpoints:
    - num: 1
      device_type: COLOR_DIMMABLE_LIGHT
      clusters:
        - id: ON_OFF
          attributes:
            - id: 0
              type: bool
              on_value:
                then:
                  - light.control:
                      id: light_1
                      state: !lambda "return (bool)x;"
        - id: LEVEL_CONTROL
          attributes:
            - id: 0
              type: U8
              on_value:
                then:
                  - light.control:
                      id: light_1
                      brightness: !lambda "return ((float)x)/255;"
        - id: COLOR_CONTROL
          attributes:
            - id: 3
              type: U16
              on_value:
                then:
                  - lambda: id(color_x) = (float)x/65536;
                  - light.control:
                      id: light_1
                      red: !lambda "return zigbee::get_r_from_xy(id(color_x), id(color_y));"
                      green: !lambda "return zigbee::get_g_from_xy(id(color_x), id(color_y));"
                      blue: !lambda "return zigbee::get_b_from_xy(id(color_x), id(color_y));"
            - id: 4
              type: U16
              on_value:
                then:
                  - lambda: id(color_y) = (float)x/65536;
                  - light.control:
                      id: light_1
                      red: !lambda "return zigbee::get_r_from_xy(id(color_x), id(color_y));"
                      green: !lambda "return zigbee::get_g_from_xy(id(color_x), id(color_y));"
                      blue: !lambda "return zigbee::get_b_from_xy(id(color_x), id(color_y));"
    - device_type: TEMPERATURE_SENSOR
      num: 2
      clusters:
        - id: ILLUMINANCE_MEASUREMENT
          attributes:
            - id: 0
              type: U16
              value: 1
```

luar123, 2024-12-17:
With the current version you can use this setup:
```yaml
sensor:
  - platform: bh1750
    id: ilum
    name: "BH1750 Illuminance"
    address: 0x23
    update_interval: 60s
    filters:
      - delta: 5
zigbee:
  id: "zb"
  endpoints:
    - device_type: LIGHT_SENSOR
      num: 1
      clusters:
        - id: ILLUMINANCE_MEASUREMENT
          attributes:
            - attribute_id: 0
              type: U16
              report: true
              value: 0
              lambda: !lambda "return log10(x)*10000 + 1;"
              device: ilum
```
Of course you can add more clusters/endpoints/attributes.

closed 2024-12-17 by luar123