luar123/zigbee_esphome #5: Zigbee attribute reporting

luar123, 2024-06-23:
This came up in #4:

> But I found an issue where the delay in sensor data reporting is very high, and it can take from one minute to a few minutes to wait. It can be reported normally in your esphome_zb_sensor project.

There are multiple ways of attribute reporting available:
1. you can issue a report command, to report the attribute value immediately. This is done with `zigbee.report` for all attributes that are written with `zigbee.setAttr `
2. you can configure reporting based on value change or time interval. This is activated by default for all attributes that are written with `zigbee.setAttr ` with setting: min_interval = 10, max_interval = 0, delta = 0 -> report on every attribute change, but no periodic reporting. This setting is hardcoded at the moment, but can be overwritten by the coordinator (e.g. zigbee2mqtt)

esphome_zb_sensor uses the brute-force method: Whenever the attribute is set, a report command is issued. If you want to replicate this (I won't recommend) run `zigbee.report` after `zigbee.setAttr `.

**Important:** If you use z2m with the default settings and either no custom configuration for your device or using the modern extend functions, reporting settings might get overwritten. E.g. for temperature it is set to an interval between 10s and 1h and a difference of 1°C: https://github.com/Koenkk/zigbee-herdsman-converters/blob/41b26f2025b0627f984ccf0da65ca4051bd404f0/src/lib/modernExtend.ts#L561
You can always change these settings through the web frontend.

papylhomme, 2024-09-22:
Hello, I stumbled onto this issue before opening a new one. For me reporting doesn't seem to work at all, I even added an interval to call the `zigbee.report` action with no luck. I'm using HA with  ZHA (coordinator is a SLZB-06M).

The clusters are configured with `report: true`, and the interval component is as follow (I can see the execution in the logs)

```
interval:
  - interval: 1min
    then:
      - logger.log: "Zigbee Report !"
      - zigbee.report: zb
```

For each report call, there is a Zigbee warning/error from 30s to 1m later :
`[15:19:21][I][zigbee:193][Zigbee_main]: ZDO signal: ZDO Device Unavailable (0x3c), status: ESP_OK`

However I can successfully read the attributes directly by issuing a command from HA. 
[image]

Any idea how I can investigate this further ?

luar123, 2024-11-02:
Could you solve this in the meantime?
If not, you could try to get more info from ZHA debug logs or capture the zigbee packets with wireshark: https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html
Or use z2m. Maybe there is some compatibility issue with ZHA (I never tried).

patrickfnielsen, 2024-12-16:
Hey - I'm not sure if I should post in this issue or create a new one, but I'll start here.

I'm also having trouble getting reporting to work (I use Z2M). I have successfully added the device to Z2M, and Z2M discovers the need clusters/attributes without any issue.

Pressing the "refresh" button, or reading the attributes work, but when I update a value with `zigbee.setAttr` nothing is reported, though I can see it setting the attribute in the log. The reporting tab in Z2M looks like I would expect (using modern extent).

If I add a `zigbee.report: zb` as the second action (see example) it does report the value to Z2M, but then I would have to do all filters in the esphome config (delta, etc), and it seems like a workaround?

```yaml
zigbee:
  id: "zb"
  endpoints:
    - num: 1
      device_type: SIMPLE_SENSOR
      clusters:
        - id: OCCUPANCY_SENSING
          attributes:
            - id: 0x0000
              type: 8BITMAP
              report: true
            - id: 0x0001
              type: 8BIT_ENUM
              value: 0x0
            - id: 0x0002
              type: 8BITMAP
              value: 0x0

        - id: ILLUMINANCE_MEASUREMENT
          attributes:
            - id: 0x0000
              type: U16
              report: true
            - id: 0x0004
              type: U16
              value: 0x0

binary_sensor:
  - platform: LD2412
    has_target:
      name: Presence
      on_state:
        then:
          - zigbee.setAttr:
              id: zb
              endpoint: 1
              cluster: OCCUPANCY_SENSING
              attribute: 0x0000
              value: !lambda "return x;"
          - zigbee.report: zb

sensor:
  - platform: bh1750
    name: Illuminance
    address: 0x23
    update_interval: 10s
    filters: 
      - delta: 8
    on_value: 
      then:
        - zigbee.setAttr:
            id: zb
            endpoint: 1
            cluster: ILLUMINANCE_MEASUREMENT
            attribute: 0x0000
            value: !lambda "return x;"
        - zigbee.report: zb
```

I'm not sure where to go frome here, so any help would be appreciated.

luar123, 2024-12-16:
Try to add `value: 0` to Attribute 0 of illuminance cluster.

> If I add a zigbee.report: zb as the second action (see example) it does report the value to Z2M, but then I would have to do all filters in the esphome config (delta, etc), and it seems like a workaround?

Yes, that is only a workaround and would use another reporting mechanism.

Is there anything in the logs?

patrickfnielsen, 2024-12-16:
I just updated to the new syntax (good job with that btw), and get the same behavior as before (with `value: 0`added):
```yaml
zigbee:
  id: "zb"
  endpoints:
    - num: 1
      device_type: SIMPLE_SENSOR
      clusters:
        - id: OCCUPANCY_SENSING
          attributes:
            - attribute_id: 0x0000
              type: 8BITMAP
              report: true
              device: occupancy
            - attribute_id: 0x0001
              type: 8BIT_ENUM
              value: 0x0
            - attribute_id: 0x0002
              type: 8BITMAP
              value: 0x0

        - id: ILLUMINANCE_MEASUREMENT
          attributes:
            - attribute_id: 0x0000
              type: U16
              value: 0
              report: true
              device: illuminance
            - attribute_id: 0x0004
              type: U16
              value: 0x0

sensor:
  - platform: bh1750
    id: illuminance
    name: Illuminance
    address: 0x23
    update_interval: 10s
    filters: 
      - delta: 8

binary_sensor:
  - platform: LD2412
    has_target:
      id: occupancy
      name: Presence
```

Here's the relevant zigbee logs (I think):
```
[10:50:35][D][esp-idf:000]: E (291) ESP_ZIGBEE_ATTRIBUTE: The requested update attribute ID:0x10 is not found
[10:50:35][D][esp-idf:000]: E (292) ESP_ZIGBEE_ATTRIBUTE: The requested update attribute ID:0x4 is not found
[10:50:36][D][esp-idf:000][Zigbee_main]: E (1153) ESP_ZIGBEE_CLUSTER: The requested update cluster ID (0x   0) is not found
[10:50:36][D][esp-idf:000][Zigbee_main]: E (1154) zigbee_helper: Ignore previous cluster not found error
[10:50:36][D][esp-idf:000][Zigbee_main]: E (1155) ESP_ZIGBEE_CLUSTER: The requested update cluster ID (0x 400) is not found
[10:50:36][D][esp-idf:000][Zigbee_main]: E (1155) zigbee_helper: Ignore previous cluster not found error
[10:50:36][D][esp-idf:000][Zigbee_main]: E (1156) ESP_ZIGBEE_CLUSTER: The requested update cluster ID (0x 406) is not found
[10:50:36][D][esp-idf:000][Zigbee_main]: E (1157) zigbee_helper: Ignore previous cluster not found error
[10:50:36][I][zigbee:356][Zigbee_main]: set reporting for cluster: 1030
[10:50:36][I][zigbee:356][Zigbee_main]: set reporting for cluster: 1024
[10:50:36][I][zigbee:182][Zigbee_main]: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
[10:50:36][I][zigbee:094][Zigbee_main]: SKIP_STARTUP. Device started up in non factory-reset mode
[10:50:36][I][zigbee:096][Zigbee_main]: Initialize Zigbee stack
[10:50:37][D][sensor:093]: 'Illuminance': Sending state 57.95275 lx with 1 decimals of accuracy
[10:50:37][D][zigbee:019]: Attribute set!
[10:50:37][C][zigbee:407]: ZigBee:
[10:50:37][C][zigbee:409]: Endpoint: 1, 12
[10:50:38][D][binary_sensor:036]: 'Presence': Sending state OFF
[10:50:38][D][zigbee:019]: Attribute set!
[10:50:38][I][zigbee:116][Zigbee_main]: Start network steering after reboot
[10:50:38][I][zigbee:133][Zigbee_main]: Joined network successfully (Extended PAN ID: 82:5d:19:42:3d:96:d6:ac, PAN ID: 0xbc7c, Channel:15)
[10:50:38][D][main:109][Zigbee_main]: Joined network
[10:50:38][D][zigbee:080][Zigbee_main]: Bind response from address(0x0), endpoint(1) with status(0)
[10:50:38][D][zigbee:080][Zigbee_main]: Bind response from address(0x0), endpoint(1) with status(0)
[10:52:12][D][binary_sensor:036]: 'Presence': Sending state ON
[10:52:12][D][zigbee:019]: Attribute set!
[10:52:12][D][bh1750.sensor:159]: 'Illuminance': Got illuminance=2.8lx
[10:52:12][D][sensor:093]: 'Illuminance': Sending state 2.82972 lx with 1 decimals of accuracy
[10:52:12][D][zigbee:019]: Attribute set!
[10:52:22][D][bh1750.sensor:159]: 'Illuminance': Got illuminance=60.6lx
[10:52:22][D][sensor:093]: 'Illuminance': Sending state 60.55610 lx with 1 decimals of accuracy
[10:52:22][D][zigbee:019]: Attribute set!
```

I also see this in the logs at random times:
```
[10:54:01][I][zigbee:182][Zigbee_main]: ZDO signal: ZDO Device Unavailable (0x3c), status: ESP_OK

```

luar123, 2024-12-16:
For illuminance:
I would guess reporting works as intended, but the delta is lower than the threshold configured by z2m. Check the reporting tab in z2m frontend (min rep change).

The main issue is that the zigbee attribute value is not in lux. You need to add a lambda `10000*log10(x)+1`. (Your lux value in z2m is probably close to 0.)

I have the same setup working fine. Will add my config once I get back on my PC.

I would also change simple_sensor to light_sensor.

luar123, 2024-12-17:
Thanks for the update/logs.
I posted my config [here](https://github.com/luar123/zigbee_esphome/issues/4#issuecomment-2549548748), but I guess it won't help in your case.

There seems to be a setup specific issue, I also saw some discussions about this in arduino issue reports. You are right `[Zigbee_main]: Receive Zigbee default response callback` always shows up after reporting, but in your case there seems to be a huge delay.

I will try to investigate and see how they solved it in arduino. Could you provide more information?
* full esp log from boot, z2m logs with debug log level
* full yaml
* hardware setup: coordinator/other devices/routers? You could add a screenshot from the z2m frontend map
* screenshots from z2m frontend: details/bindings/reports pages

(please remove any sensitive information)

patrickfnielsen, 2024-12-18:
Coordinator: Sonoff Dongle-P:
Coordinator type: zStack3x0
Coordinator revision: 20230507 

Full ESP config:
```yaml
esphome:
  name: zigbeetest
  friendly_name: ZigbeeTest

esp32:
  board: esp32-c6-devkitc-1
  flash_size: 4MB
  partitions: partitions_zb.csv
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_4MB: y

external_components:
  - source: github://luar123/zigbee_esphome
    components: [ zigbee ]

logger:
  level: DEBUG
  hardware_uart: USB_SERIAL_JTAG

i2c:
  sda: GPIO22
  scl: GPIO23
  scan: false

zigbee:
  id: "zb"
  endpoints:
    - num: 1
      device_type: LIGHT_SENSOR
      clusters:
        - id: ILLUMINANCE_MEASUREMENT
          attributes:
            - id: illuminance_attr
              attribute_id: 0x0000
              type: U16
              value: 0x0
              report: true
              #device: illuminance
            - attribute_id: 0x0004
              type: U16
              value: 0x0

  on_join:
    then:
      - logger.log: "Joined network"

sensor:
  - platform: bh1750
    id: illuminance
    name: Illuminance
    address: 0x23
    update_interval: 10s
    on_value: 
      then:
        - zigbee.setAttr:
            id: illuminance_attr
            value: !lambda "return 10000*log10(x)+1;"
        #- zigbee.report: zb
```

ESP32-C6 boot logs:
```
[... 137 lines cut ...]
[2024-12-18 10:21:37] debug: 	zh:controller:endpoint: ZCL command 0x543204fffe215ca4/1 msIlluminanceMeasurement.defaultRsp({"cmdId":10,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"transactionSequenceNumber":5,"writeUndiv":false})
[2024-12-18 10:21:37] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0x543204fffe215ca4:62258/1 (0,0,1)
[2024-12-18 10:21:37] debug: 	zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":62258,"destendpoint":1,"srcendpoint":1,"clusterid":1024,"transid":123,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,5,11,10,0]}}
[2024-12-18 10:21:37] debug: 	zh:zstack:unpi:writer: --> frame [254,15,36,1,50,243,1,1,0,4,123,0,30,5,24,5,11,10,0,147]
[2024-12-18 10:21:37] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-12-18 10:21:37] debug: 	z2m: Received Zigbee message from '0x543204fffe215ca4', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":3963}' from endpoint 1 with groupID 0
[2024-12-18 10:21:37] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x543204fffe215ca4', payload '{"illuminance":3963,"illuminance_lux":2,"linkquality":69}'
[2024-12-18 10:21:37] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2024-12-18 10:21:37] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2024-12-18 10:21:37] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2024-12-18 10:21:37] debug: 	zh:zstack:znp: <-- SRSP: AF - dataRequest - {"status":0}
[2024-12-18 10:21:37] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-12-18 10:21:37] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,123,189]
[2024-12-18 10:21:37] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,123,189]
[2024-12-18 10:21:37] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,123] - 189
[2024-12-18 10:21:37] debug: 	zh:zstack:znp: <-- AREQ: AF - dataConfirm - {"status":0,"endpoint":1,"transid":123}
[2024-12-18 10:21:37] debug: 	zh:zstack:unpi:parser: --- parseNext []
```

Screenshots (sorry about the snow, forgot to disable it for the screenshots):
[image]
[image]
[image]
[image]
[image]

luar123, 2024-12-20:
Thanks, I will look into this, but could take a while.

luar123, 2024-12-24:
Please try the latest commit, might or might not help. I don't really know whats wrong, but there are a couple of things we could try.
When testing, please remove the device from z2m and restart z2m and coordinator.

papylhomme, 2024-12-26:
> Please try the latest commit, might or might not help. I don't really know whats wrong, but there are a couple of things we could try. When testing, please remove the device from z2m and restart z2m and coordinator.

Hello, I did some tests with Z2M some months ago but the results weren't better than ZHA.
Right now I'm playing with your last changes on a test network with Z2M and it seems really promising for my use case :
reporting set to min:0, max:10, change:0 => I get the report for every sensor update :)

Coordinator: SMLIGHT SLZB-07 - type EZSP v12

Config:
```yaml
sensor:
  - platform: bme280_i2c
    address: 0x76
    update_interval: 10s
    temperature:
      id: bme280_temperature
      name: "BME280 Temperature"
      oversampling: 16x
    humidity:
      id: bme280_humidity
      name: "BME280 Humidity"
    pressure:
      id: bme280_pressure
      name: "BME280 Pressure"

zigbee:
  id: "zb"
  endpoints:
    - device_type: TEMPERATURE_SENSOR
      num: 1
      clusters:
        - id: REL_HUMIDITY_MEASUREMENT
          attributes:
            - attribute_id: 0
              type: U16
              report: true
              value: 200
              device: bme280_humidity
              scale: 100
        - id: TEMP_MEASUREMENT
          attributes:
            - attribute_id: 0x0
              type: S16
              report: true
              value: 100
              device: bme280_temperature
              scale: 100
        - id: PRESSURE_MEASUREMENT
          attributes:
            - attribute_id: 0x0
              type: S16
              report: true
              value: 100
              device: bme280_pressure
```

ESP Logs:
```
[09:33:39][D][sensor:093]: 'BME280 Temperature': Sending state 15.81965 °C with 1 decimals of accuracy
[09:33:39][D][zigbee:019]: Attribute set!
[09:33:39][D][sensor:093]: 'BME280 Pressure': Sending state 1010.58234 hPa with 1 decimals of accuracy
[09:33:39][D][zigbee:019]: Attribute set!
[09:33:39][D][sensor:093]: 'BME280 Humidity': Sending state 44.47461 % with 1 decimals of accuracy
[09:33:39][D][zigbee:019]: Attribute set!
[09:33:39][D][zigbee:264][Zigbee_main]: Receive Zigbee default response callback
[09:33:39][D][zigbee:264][Zigbee_main]: Receive Zigbee default response callback
[09:33:39][D][zigbee:264][Zigbee_main]: Receive Zigbee default response callback
[09:33:49][D][sensor:093]: 'BME280 Temperature': Sending state 15.80949 °C with 1 decimals of accuracy
[09:33:49][D][zigbee:019]: Attribute set!
[09:33:49][D][sensor:093]: 'BME280 Pressure': Sending state 1010.57843 hPa with 1 decimals of accuracy
[... 122 lines cut ...]
info 2024-12-26 10:33:59z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x744dbdfffe63761a', payload '{"humidity":44.23,"linkquality":255,"pressure":101,"temperature":15.8}'
info 2024-12-26 10:34:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x744dbdfffe63761a', payload '{"humidity":44.23,"linkquality":255,"pressure":101,"temperature":15.92}'
info 2024-12-26 10:34:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x744dbdfffe63761a', payload '{"humidity":54.06,"linkquality":255,"pressure":101,"temperature":15.92}'
info 2024-12-26 10:34:19z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x744dbdfffe63761a', payload '{"humidity":54.06,"linkquality":255,"pressure":101,"temperature":15.93}'
info 2024-12-26 10:34:19z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x744dbdfffe63761a', payload '{"humidity":45.05,"linkquality":255,"pressure":101,"temperature":15.93}'
info 2024-12-26 10:34:29z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x744dbdfffe63761a', payload '{"humidity":45.05,"linkquality":255,"pressure":101,"temperature":16}'
info 2024-12-26 10:34:29z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x744dbdfffe63761a', payload '{"humidity":52.77,"linkquality":255,"pressure":101,"temperature":16}'
info 2024-12-26 10:34:39z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x744dbdfffe63761a', payload '{"humidity":45.92,"linkquality":255,"pressure":101,"temperature":16}'
info 2024-12-26 10:34:49z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x744dbdfffe63761a', payload '{"humidity":45.92,"linkquality":255,"pressure":101,"temperature":16.07}'
info 2024-12-26 10:34:49z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x744dbdfffe63761a', payload '{"humidity":48.74,"linkquality":255,"pressure":101,"temperature":16.07}'
info 2024-12-26 10:34:59z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x744dbdfffe63761a', payload '{"humidity":48.74,"linkquality":255,"pressure":101,"temperature":18.53}'
info 2024-12-26 10:34:59z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x744dbdfffe63761a', payload '{"humidity":86.29,"linkquality":255,"pressure":101,"temperature":18.53}'
info 2024-12-26 10:35:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x744dbdfffe63761a', payload '{"humidity":86.29,"linkquality":255,"pressure":101,"temperature":17.91}'
info 2024-12-26 10:35:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x744dbdfffe63761a', payload '{"humidity":87.71,"linkquality":255,"pressure":101,"temperature":17.91}'
info 2024-12-26 10:35:19z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x744dbdfffe63761a', payload '{"humidity":87.71,"linkquality":255,"pressure":101,"temperature":16.77}'
info 2024-12-26 10:35:19z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x744dbdfffe63761a', payload '{"humidity":64.45,"linkquality":255,"pressure":101,"temperature":16.77}'
[... cut ...]
info 2024-12-26 10:35:39z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x744dbdfffe63761a', payload '{"humidity":45.91,"linkquality":255,"pressure":101,"temperature":16.53}'
info 2024-12-26 10:35:49z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x744dbdfffe63761a', payload '{"humidity":45.91,"linkquality":255,"pressure":101,"temperature":16.43}'
info 2024-12-26 10:35:49z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x744dbdfffe63761a', payload '{"humidity":45.01,"linkquality":255,"pressure":101,"temperature":16.43}'
```

I'll try to update one a my "production" board to see if it works with ZHA too.

patrickfnielsen, 2025-01-05:
Sorry for the delayed response.

I just tried again using the new commit, but I still see the same pattern.
To see if it was my zigbee network I also tried coding it using the ESP32 Arduino Zigbee SDK, and everything works as when I use that with the same setup.

luar123, 2025-01-06:
I want to update to the latest zigbee sdk 1.6.1 (same as arduino uses). If that doesn't help you could send me your Arduino code, then I can compare.

luar123, 2025-01-11:
@patrickfnielsen: please try esp sdk 1.6.1 version.
As always `esphome clean`, `esptool.py erase_flash` and re-interview and remove/add device in z2m is needed.

patrickfnielsen, 2025-01-19:
I tried the latest version, but I cant seem to get over a panic now:
```
[17:17:11][D][bh1750.sensor:159]: 'Illuminance': Got illuminance=95.5lx
[17:17:11][D][sensor:093]: 'Illuminance': Sending state 95.53149 lx with 1 decimals of accuracy
[17:17:11]Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.
```

It seems to happen after sending. Same YAML as before. 
I did a clean, erase_flash, restarted Z2M + coordinator, but still get crashes everytime

luar123, 2025-01-19:
Sorry, forgot to test the SetAttr action with the latest changes. Should work now.

patrickfnielsen, 2025-01-22:
@luar123 I did another test, and it seems to work as expected now! Thanks a lot for the help, and the great project!

papylhomme, 2025-01-27:
Updated to the last version and everything is working fine too 👍

luar123, 2025-03-09:
As always everything can be found in ZCL: type is float not integer, max is 1 so you probably need to set the max value attribute. See pm2.5 cluster here: https://github.com/luar123/zigbee_esphome/discussions/22#discussioncomment-11881954
You might need to check if z2m expects a ppm value at all.

Please create a new issue if still needed.

closed 2025-03-09 by luar123