luar123/zigbee_esphome #8: Examples seem not to log properly

fellnerse, 2024-08-17:
Hey, I'm using your c6 examples.
first of all, really awesome, that you created this esphome integration for idf. One thing, that seems not to be working for me is the logging. I only get this:
```
[20:33:33]I (454) sleep: Configure to isolate all GPIO pins in sleep state
[20:33:33]I (461) sleep: Enable automatic switching of GPIO sleep configuration
[20:33:33]I (468) coexist: coex firmware version: 77cd7f8
[20:33:33]I (473) coexist: coexist rom version 5b8dcfa
[20:33:33]I (478) main_task: Started on CPU0
[20:33:33]I (481) main_task: Calling app_main()
```
But nothing else is logged. Do you have any idea what I have to change? I played already with the log levels etc.

If I have a look at the `logger_esp32.cpp` file in the build/esphome/components I can see that `USE_ESP32` is not defined, not sure if that has something to do with it.

luar123, 2024-08-24:
I don't have a C6 board to test, but you could try the different options for `hardware_uart: UART0`. `UART0 `setting is supposed to be used with the USB connector labeled `UART`, not `USB`.

fellnerse, 2024-08-24:
Thank you for your quick answer!

Sorry for those noob questions, I just started with developing with idf.

You are right, it has to have something to do with that. Unfortunately, I only have a board with only one usb header, so I suppose that's the wrong one. Do you have an idea, how I could still get access to that output via the usb header? When I just only plain idf code I can log with the standard logging Funktion to the console. It seems to be related to esphome, as an empty hello world also does not work. 

If you have any resources I could dig deeper I would appreciate it much. I can also create a mr, locally I was able to upgrade to idf 5.3. It's very cool that you provide this integration!

luar123, 2024-08-24:
I think in that case you have to set `hardware_uart` to `USB_CDC` or `USB_SERIAL_JTAG`. Not sure if this is fully supported by esphome as C6 and IDF5 are officially not supported. More info here: https://esphome.io/components/logger#hardware-uarts
Anyway, this is not related to the zigbee integration, so I will close here. Checkout esphome on discord to get support.

closed 2024-08-24 by luar123