luar123/zigbee_esphome #9: Look into if could make use of new Zigbee Wrapper library and API for Arduino?

Hedda, 2024-08-27:
Espressif developer @P-R-O-C-H-Y is now working on a new Zigbee Wrapper library with a simpler API in Arduino (for arduino-esp32) and he wrote that other developers can post additional feature request related to that in the same issue there for tracking. Maybe you could make use of the same Zigbee API?

* https://github.com/espressif/arduino-esp32/issues/10135

luar123, 2024-08-28:
As soon as this supports all functionality that could be an option. On the other hand I would prefer to not rely on arduino too much. I believe esphome wants to support more functionality without arduino libs.

Hedda, 2024-09-06:
FYI, sounds like the initial Zigbee API / library might be merged soon into Espressif's Arduino core for ESP32 (arduino-esp32):

* https://github.com/espressif/arduino-esp32/pull/10265