luar123/zigbee_esphome #10: Network steering was not successful

rvalitov, 2024-10-26:
Hi!
Thank you for such a great project! I'm trying to build the [example](https://github.com/luar123/zigbee_esphome/blob/master/example_esp32c6.yaml) on ESP32-C6 but I get error that `Network steering was not successful`.
The device does not connect to Z2M in HA. Any advice?

```
[15:45:28]I (23) boot: ESP-IDF 5.2.1 2nd stage bootloader
[15:45:28]I (24) boot: compile time Oct 26 2024 15:34:03
[15:45:28]I (24) boot: chip revision: v0.1
[15:45:28]I (26) boot.esp32c6: SPI Speed      : 80MHz
[15:45:28]I (31) boot.esp32c6: SPI Mode       : DIO
[15:45:28]I (36) boot.esp32c6: SPI Flash Size : 4MB
[15:45:28]I (41) boot: Enabling RNG early entropy source...
[15:45:28]I (46) boot: Partition Table:
[15:45:28]I (50) boot: ## Label            Usage          Type ST Offset   Length
[15:45:28]I (57) boot:  0 otadata          OTA data         01 00 00009000 00002000
[15:45:28]I (64) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[15:45:28]I (72) boot:  2 app0             OTA app          00 10 00010000 001b0000
[15:45:28]I (79) boot:  3 app1             OTA app          00 11 001c0000 001b0000
[15:45:28]I (87) boot:  4 nvs              WiFi data        01 02 00370000 0006d000
[15:45:28]I (94) boot:  5 zb_storage       Unknown data     01 81 003dd000 00004000
[15:45:28]I (102) boot:  6 zb_fct           Unknown data     01 81 003e1000 00000400
[15:45:28]I (109) boot: End of partition table
[15:45:28]I (114) boot: No factory image, trying OTA 0
[15:45:28]I (118) esp_image: segment 0: paddr=00010020 vaddr=42080020 size=12ad8h ( 76504) map
[15:45:28]I (143) esp_image: segment 1: paddr=00022b00 vaddr=40800000 size=0d344h ( 54084) load
[15:45:28]I (156) esp_image: segment 2: paddr=0002fe4c vaddr=4080d350 size=001cch (   460) load
[15:45:28]I (157) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=78630h (493104) map
[15:45:28]I (264) esp_image: segment 4: paddr=000a8658 vaddr=4080d51c size=01b1ch (  6940) load
[15:45:28]I (269) boot: Loaded app from partition at offset 0x10000
[15:45:28]I (327) boot: Set actual ota_seq=1 in otadata[0]
[15:45:28]I (327) boot: Disabling RNG early entropy source...
[15:45:28]I (338) cpu_start: Unicore app
[15:45:28]W (346) clk: esp_perip_clk_init() has not been implemented yet
[15:45:28]I (353) cpu_start: Pro cpu start user code
[15:45:28]I (353) cpu_start: cpu freq: 160000000 Hz
[15:45:28]I (353) cpu_start: Application information:
[15:45:28]I (356) cpu_start: Project name:     zb-example-c6
[15:45:28]I (361) cpu_start: App version:      2024.10.1
[15:45:28]I (367) cpu_start: Compile time:     Oct 26 2024 15:31:50
[15:45:28]I (373) cpu_start: ELF file SHA256:  d05fd6d24...
[15:45:28]I (378) cpu_start: ESP-IDF:          5.2.1
[15:45:28]I (383) cpu_start: Min chip rev:     v0.0
[15:45:28]I (388) cpu_start: Max chip rev:     v0.99
[15:45:28]I (392) cpu_start: Chip rev:         v0.1
[15:45:28]I (397) heap_init: Initializing. RAM available for dynamic allocation:
[15:45:28]I (404) heap_init: At 40813990 len 00068C80 (419 KiB): RAM
[15:45:28]I (410) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
[15:45:28]I (417) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
[15:45:28]I (424) spi_flash: detected chip: generic
[15:45:28]I (428) spi_flash: flash io: dio
[15:45:28]W (431) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
[15:45:28]W (445) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`
[15:45:28]I (456) sleep: Configure to isolate all GPIO pins in sleep state
[15:45:28]I (462) sleep: Enable automatic switching of GPIO sleep configuration
[15:45:28]I (469) coexist: coex firmware version: 77cd7f8
[15:45:28]I (475) coexist: coexist rom version 5b8dcfa
[15:45:28]I (480) main_task: Started on CPU0
[15:45:28]I (483) main_task: Calling app_main()
[15:45:28]I (588) main_task: Returned from app_main()
[15:45:28][I][logger:156]: Log initialized
[15:45:28][I][zigbee:269]: Model: zb-example-c6
[15:45:28][I][zigbee:270]: Manufacturer: esphome
[15:45:28][I][zigbee:271]: Date: 20241026
[15:45:28][I][zigbee:272]: Area:
[15:45:28][I][zigbee:269]: Model: zb-example-c6
[15:45:28][I][zigbee:270]: Manufacturer: esphome
[15:45:28][I][zigbee:271]: Date: 20241026
[15:45:28][I][zigbee:272]: Area:
[15:45:28][I][app:029]: Running through setup()...
[15:45:28][C][esp32_rmt_led_strip:021]: Setting up ESP32 LED Strip...
[15:45:28][D][esp-idf:000]: I (616) gpio: GPIO[9]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
[15:45:28]
[15:45:28][D][binary_sensor:034]: 'button_1': Sending initial state OFF
[15:45:28][C][light:035]: Setting up light 'light_1'...
[15:45:28][D][light:036]: 'light_1' Setting:
[15:45:28][D][light:041]:   Color mode: RGB
[15:45:28][D][light:085]:   Transition length: 1.0s
[15:45:28][D][esp-idf:000][Zigbee_main]: I (658) phy_init: phy_version 250,e14681b,Jan 24 2024,17:43:11
[15:45:28]
[15:45:28][D][esp-idf:000][Zigbee_main]: I (727) phy: libbtbb version: 939f79c, Jan 24 2024, 17:43:26
[15:45:28]
[15:45:28][D][esp-idf:000][Zigbee_main]: E (728) ESP_ZIGBEE_CLUSTER: The requested update cluster ID (0x 405) is not found
[15:45:28]
[15:45:28][I][zigbee:346][Zigbee_main]: set reporting for cluster: 1029
[15:45:28][I][zigbee:346][Zigbee_main]: set reporting for cluster: 1026
[15:45:28][I][zigbee:193][Zigbee_main]: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
[15:45:28][I][zigbee:105][Zigbee_main]: SKIP_STARTUP. Device started up in factory-reset mode
[15:45:28][I][zigbee:107][Zigbee_main]: Initialize Zigbee stack
[15:45:29][I][zigbee:113][Zigbee_main]: FIRST_START. Device started up in factory-reset mode
[15:45:29][I][zigbee:116][Zigbee_main]: Start network steering
[15:45:29][C][internal_temperature:076]: Internal Temperature Sensor 'Internal Temperature'
[15:45:29][C][internal_temperature:076]:   Device Class: 'temperature'
[15:45:29][C][internal_temperature:076]:   State Class: 'measurement'
[15:45:29][C][internal_temperature:076]:   Unit of Measurement: '▒C'
[15:45:29][C][internal_temperature:076]:   Accuracy Decimals: 1
[15:45:31][I][zigbee:172][Zigbee_main]: Network steering was not successful (status: ESP_FAIL)
[15:45:34][I][zigbee:172][Zigbee_main]: Network steering was not successful (status: ESP_FAIL)
[15:45:37][I][zigbee:172][Zigbee_main]: Network steering was not successful (status: ESP_FAIL)
[15:45:40][I][zigbee:172][Zigbee_main]: Network steering was not successful (status: ESP_FAIL)
[15:45:44][I][zigbee:172][Zigbee_main]: Network steering was not successful (status: ESP_FAIL)
[15:45:47][I][zigbee:172][Zigbee_main]: Network steering was not successful (status: ESP_FAIL)
[15:45:50][I][zigbee:172][Zigbee_main]: Network steering was not successful (status: ESP_FAIL)
[15:45:53][I][zigbee:172][Zigbee_main]: Network steering was not successful (status: ESP_FAIL)

```

luar123, 2024-10-26:
I can't see something obvious wrong.
Joining in z2m is enabled and the device is in close to the coordinator? Check z2m debug log if there is something.
Did you use the example as is?
Otherwise try to delete your build folder, erase the whole flash, if you have a experimental z2m setup you could stop z2m, disconnect the coordinator, delete state.json, database.db and coordinator_backup.json.

rvalitov, 2024-10-26:
Thank you for a prompt response! I will try to check the Z2m logs. The device should go into pairing mode automatically? Or I need to hold the button? I tried that in any case and there's no difference. Anyway I would like to know the pairing logic.

luar123, 2024-10-26:
After first boot or after `zigbee.reset` (pressing the button for >5s) the device tries to join a network 10 times with 1s break and then every 10 minutes until it succeeds.
Every `Network steering was not successful` is a failed attempt to find or join a network.

rvalitov, 2024-10-29:
I made some tests. The reason was that the device was too far from the coordinator. It's just behind a thin 4-5 cm wall. For all other devices in this room I have about 170 LQI. So I never thought that the signal strength of this ESP32-C6 is so surprisingly low. Perhaps, there's a way to change the power in the config? Does the H2 also has so poor Zigbee radio strength?

closed 2024-10-29 by rvalitov

luar123, 2024-10-31:
My H2 shows 118lqi, 50 cm from an old CC2531 usb dongle coordinator. Don't have other devices connected to compare...
Could you flash a pure esp-idf example to see if it is the same? I don't have a C6 for testing here...

rvalitov, 2024-10-31:
What do you mean by pure example? Can you share it?

luar123, 2024-10-31:
See readme. E.g https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_on_off_light

rvalitov, 2024-11-01:
OK, thanks for the link! So I should compare just the LQI level?

dvv, 2025-01-15:
@rvalitov Hi! Have you finally found the solution?

rvalitov, 2025-01-28:
@dvv yes I just had to put the device to the same room as coordinator.