luar123/zigbee_esphome #11: Help with ELECTRICAL_MEASUREMENT

remy, 2024-10-27:
I've been trying to find a zigbee solution to measuring an external battery source (in my case, it'll be a 6v source down to 3v using a voltage divider) that sleeps every few hours and then reports again.

I've got the `example_esp32h2.yaml` to build and flash, and I can control the device through Home Assistant (i.e. to turn the LED on, change colour, etc).

I've started adding the following under the `zigbee.endpoints` path in the yaml file:

```yaml
    - num: 3
      device_type: SIMPLE_SENSOR
      clusters:
        - id: ELECTRICAL_MEASUREMENT
          attributes:
            - id: 0x0
              type: U32
              report: true
              value: 6
            - id: 0x01
              type: S16
              report: true
              value: 0x8000
```

But this causes the ESP to panic crash once flashed (and I'm not entirely sure how to properly tell the esphome to read an external voltage from the pin and "wire" it through yaml to report as a DC voltage - so I very much expect my changes are duff anyway).

This is the panic + dump:

```
[13:10:26]ESP_ERROR_CHECK failed: esp_err_t 0x101 (ESP_ERR_NO_MEM) at 0x42007964
[13:10:26]file: "src/esphome/components/zigbee/zigbee.cpp" line 354
[13:10:26]func: void esphome::zigbee::ZigBeeComponent::esp_zb_task()
[13:10:26]expression: esp_zb_zcl_update_reporting_info(&reporting_info)
[13:10:26]
[13:10:26]abort() was called at PC 0x408060e7 on core 0
[13:10:26]Core  0 register dump:
[13:10:26]MEPC    : 0x40801784  RA      : 0x408060f2  SP      : 0x4081b230  GP      : 0x4080bf60  
WARNING Decoded 0x40801784: panic_abort at /Users/remy/.platformio/packages/framework-espidf/components/esp_system/panic.c:466
WARNING Decoded 0x40801784: panic_abort at /Users/remy/.platformio/packages/framework-espidf/components/esp_system/panic.c:466
```

There's more with a full register dump, but I'm not sure that would be useful.

Can you offer any guidance as to how I could add DC voltage tracking to this device? (thanks in advance)

luar123, 2024-10-27:
Not all clusters/attributes support reporting, check the official cluster library. Or simply try to remove `report: true`. Also see comment in #6

luar123, 2024-11-02:
With commit a09131d662c985b300de2c8e365c2c50cf394d06 it should not crash anymore and give proper error messages.

closed 2024-11-02 by luar123