luar123/zigbee_esphome #13: Even after deleting Zigbee end point I can see the Temperature and Humidity sensor in z2mqtt expose tab.

Hakim2014, 2024-11-16:
Hi,

No matter what I do, I cannot remove the temperature sensor from zigbee2mqtt expose tab. 

I deleted "device_type: TEMPERATURE_SENSOR" from the zigbee attribute but nothing changes at ZIgbee2MQTT coordinator. 

My ZIgbee2MQTT coordinator version is 1.37.1.

I am trying to add mmwave sensor LD2450 but no success. I can't even add the ESP CPU speed attribute  at zigbee. Please find the sensor config and zigbee config below. Also, I tried to use device type as TEMPERATURE_SENSOR itself. Please suggest what am I doing wrong.

```
globals:
  -id: color_x
  type: float
  restore_value: no
  initial_value: '0'
  -id: color_y
  type: float
  restore_value: no
  initial_value: '0'
  -id: cpu_speed
  type: int
  restore_value: no
  initial_value: '0'
  -id: last_update_ld2450
  type: unsigned long
  restore_value: no
  initial_value: '0'
  -id: init_zone_publish
  type: bool
  restore_value: no
  initial_value: "false"

sensor:
  - platform: template
    name: "ESP CPU Speed"
    accuracy_decimals: 0
    icon: mdi:cpu-32-bit
    unit_of_measurement: Mhz
    disabled_by_default: False
    on_value:
      then:
        - zigbee.setAttr:
            id: zb
            endpoint: 2
            cluster: TEMP_MEASUREMENT
            attribute: 0x0
            value: !return "(id(cpu_speed));"
    entity_category: "diagnostic"
    update_interval: 600s

zigbee:
  id: "zb"
  endpoints:
    - num: 1
      device_type: COLOR_DIMMABLE_LIGHT
      clusters:
        - id: ON_OFF
          attributes:
            - id: 0
              type: bool
              on_value:
                then:
                  - light.control:
                      id: light_1
                      state: !lambda "return (bool)x;"
        - id: LEVEL_CONTROL
          attributes:
            - id: 0
              type: U8
              on_value:
                then:
                  - light.control:
                      id: light_1
                      brightness: !lambda "return ((float)x)/255;"
        - id: COLOR_CONTROL
          attributes:
            - id: 3
              type: U16
              on_value:
                then:
                  - lambda: id(color_x) = (float)x/65536;
                  - light.control:
                      id: light_1
                      red: !lambda "return zigbee::get_r_from_xy(id(color_x), id(color_y));"
                      green: !lambda "return zigbee::get_g_from_xy(id(color_x), id(color_y));"
                      blue: !lambda "return zigbee::get_b_from_xy(id(color_x), id(color_y));"
            - id: 4
              type: U16
              on_value:
                then:
                  - lambda: id(color_y) = (float)x/65536;
                  - light.control:
                      id: light_1
                      red: !lambda "return zigbee::get_r_from_xy(id(color_x), id(color_y));"
                      green: !lambda "return zigbee::get_g_from_xy(id(color_x), id(color_y));"
                      blue: !lambda "return zigbee::get_b_from_xy(id(color_x), id(color_y));"
    - device_type: TEMPERATURE_SENSOR
      num: 2
      clusters:
        - id: TEMP_MEASUREMENT
          attributes:
            - id: 0x0
              type: S16
              report: true
              value: 100
  on_join:
    then:
      - logger.log: "Joined network"
```
Thanks for your help.

luar123, 2024-11-17:
If you want to change the endpoint/cluster config, you need at least to remove the device from z2m and join again.
Regarding your template sensor: is on_value ever triggered? You could add a log action. And not sure if your value is correct, I would try: `value: !lambda "return id(cpu_speed);"`

Hakim2014, 2024-11-18:
Thanks for your prompt reply and apologies for my limited knowledge in this area.

I have remove the temp sensor and deleted the zigbee endpoint/cluster for Temperature sensor, removed the device from zigbee2mqtt, restarted zigbee2mqtt, deployed the new Yaml to esp32 c6 board and added the device again to zigbee2mqtt. Still I can see the Temperature and Humidity entry in zigbee2Mqtt. Please find the snap shot and Yaml config below.
```
esphome:
  name: zb-example-c6

external_components:
  - source: components
    components: [ zigbee ]

esp32:
  board: esp32-c6-devkitc-1
  flash_size: 4MB
  partitions: partitions_zb.csv
  framework:
    platform_version: 6.6.0
    type: esp-idf
    version: 5.2.1
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_4MB: y
  variant: esp32c6

# Enable logging
logger:
  hardware_uart: UART0

globals:
   - id: color_x
     type: float
     restore_value: no
     initial_value: '0'
   - id: color_y
     type: float
     restore_value: no
     initial_value: '0'

zigbee:
  id: "zb"
  endpoints:
    - num: 1
      device_type: COLOR_DIMMABLE_LIGHT
      clusters:
        - id: ON_OFF
          attributes:
            - id: 0
              type: bool
              on_value:
                then:
                  - light.control:
                      id: light_1
                      state: !lambda "return (bool)x;"
        - id: LEVEL_CONTROL
          attributes:
            - id: 0
              type: U8
              on_value:
                then:
                  - light.control:
                      id: light_1
                      brightness: !lambda "return ((float)x)/255;"
        - id: COLOR_CONTROL
          attributes:
            - id: 3
              type: U16
              on_value:
                then:
                  - lambda: id(color_x) = (float)x/65536;
                  - light.control:
                      id: light_1
                      red: !lambda "return zigbee::get_r_from_xy(id(color_x), id(color_y));"
                      green: !lambda "return zigbee::get_g_from_xy(id(color_x), id(color_y));"
                      blue: !lambda "return zigbee::get_b_from_xy(id(color_x), id(color_y));"
            - id: 4
              type: U16
              on_value:
                then:
                  - lambda: id(color_y) = (float)x/65536;
                  - light.control:
                      id: light_1
                      red: !lambda "return zigbee::get_r_from_xy(id(color_x), id(color_y));"
                      green: !lambda "return zigbee::get_g_from_xy(id(color_x), id(color_y));"
                      blue: !lambda "return zigbee::get_b_from_xy(id(color_x), id(color_y));"
  on_join:
    then:
      - logger.log: "Joined network"

light:
  - platform: esp32_rmt_led_strip
    rgb_order: GRB
    pin: 8
    num_leds: 1
    rmt_channel: 0
    chipset: ws2812
    id: light_1

binary_sensor:
  - platform: gpio
    pin:
      number: 9
      mode:
        input: true
        pullup: true
      inverted: true
    id: button_1
    on_press:
      then:
        - zigbee.report: zb
    on_click:
      min_length: 5s
      max_length: 20s
      then:
        - zigbee.reset: zb
```

[image]

Thanks in advance for you help and time.

Hakim2014, 2024-11-18:
After couple of restarts and removing and adding device for at least 10 times it worked :) I could alter the cluster but now I am getting other error 
```
example_esp32c6.yaml:7:28: error: 'ESP' was not declared in this scope
*** [.pioenvs/zb-example-c6/src/main.o] Error 1
```

This is my yaml
```
esphome:
  name: zb-example-c6
  on_boot:
    - priority: 100
      then:
        lambda: |-
          id(cpu_speed) = ESP.getCpuFreqMHz();

external_components:
  - source: components
    components: [ zigbee ]

esp32:
  board: esp32-c6-devkitc-1
  flash_size: 4MB
  partitions: partitions_zb.csv
  framework:
    platform_version: 6.6.0
    type: esp-idf
    version: 5.2.1
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_4MB: y
  variant: esp32c6

# Enable logging
logger:
  hardware_uart: UART0

globals:
   - id: cpu_speed
     type: int
     restore_value: no
     initial_value: '0'
   - id: color_x
     type: float
     restore_value: no
     initial_value: '0'
   - id: color_y
     type: float
     restore_value: no
     initial_value: '0'

sensor:
  - platform: internal_temperature
    name: "Internal Temperature"
    id: "temp"
    on_value:
      then:
        - zigbee.setAttr:
            id: zb
            endpoint: 2
            cluster: TEMP_MEASUREMENT
            attribute: 0x0
            value: !lambda  "return (id(cpu_speed));"

zigbee:
  id: "zb"
  endpoints:
    - num: 1
      device_type: COLOR_DIMMABLE_LIGHT
      clusters:
        - id: ON_OFF
          attributes:
            - id: 0
              type: bool
              on_value:
                then:
                  - light.control:
                      id: light_1
                      state: !lambda "return (bool)x;"
        - id: LEVEL_CONTROL
          attributes:
            - id: 0
              type: U8
              on_value:
                then:
                  - light.control:
                      id: light_1
                      brightness: !lambda "return ((float)x)/255;"
        - id: COLOR_CONTROL
          attributes:
            - id: 3
              type: U16
              on_value:
                then:
                  - lambda: id(color_x) = (float)x/65536;
                  - light.control:
                      id: light_1
                      red: !lambda "return zigbee::get_r_from_xy(id(color_x), id(color_y));"
                      green: !lambda "return zigbee::get_g_from_xy(id(color_x), id(color_y));"
                      blue: !lambda "return zigbee::get_b_from_xy(id(color_x), id(color_y));"
            - id: 4
              type: U16
              on_value:
                then:
                  - lambda: id(color_y) = (float)x/65536;
                  - light.control:
                      id: light_1
                      red: !lambda "return zigbee::get_r_from_xy(id(color_x), id(color_y));"
                      green: !lambda "return zigbee::get_g_from_xy(id(color_x), id(color_y));"
                      blue: !lambda "return zigbee::get_b_from_xy(id(color_x), id(color_y));"
    - device_type: TEMPERATURE_SENSOR
      num: 2
      clusters:
        - id: TEMP_MEASUREMENT
          attributes:
            - id: 0x0
              type: S16
              report: true
              value: 100
  on_join:
    then:
      - logger.log: "Joined network"

light:
  - platform: esp32_rmt_led_strip
    rgb_order: GRB
    pin: 8
    num_leds: 1
    rmt_channel: 0
    chipset: ws2812
    id: light_1

binary_sensor:
  - platform: gpio
    pin:
      number: 9
      mode:
        input: true
        pullup: true
      inverted: true
    id: button_1
    on_press:
      then:
        - zigbee.report: zb
    on_click:
      min_length: 5s
      max_length: 20s
      then:
        - zigbee.reset: zb
```
Just trying to publish cpu frequency.

nliviu, 2024-12-04:
`ESP.getCpuFreqMHz()` is available only on arduino platform

closed 2025-05-25 by luar123