luar123/zigbee_esphome #15: ESP32-C6

10der, 2024-12-01:
```
[11:27:47]Rebooting...
[11:27:47]ESP-ROM:esp32c6-20220919
[11:27:47]Build:Sep 19 2022
[11:27:47]rst:0xc (SW_CPU),boot:0xc (SPI_FAST_FLASH_BOOT)
[11:27:47]Saved PC:0x4001975a
[11:27:47]SPIWP:0xee
[11:27:47]mode:DIO, clock div:2
[11:27:47]load:0x4086c410,len:0xe58
[11:27:47]load:0x4086e610,len:0x2e20
[11:27:47]load:0x40875720,len:0x1804
[11:27:47]entry 0x4086c410
```

```
[11:27:48]I (540) main_task: Started on CPU0
[11:27:48]I (543) main_task: Calling app_main()
[11:27:48]I (647) main_task: Returned from app_main()
[11:27:48]Zigbee stack assertion failed zcl/zcl_general_commands.c:608
[11:27:48]
[11:27:48]abort() was called at PC 0x42034a23 on core 0
```

10der, 2024-12-01:
the same but via Arduino
[image]
[image]

ggb83, 2024-12-29:
Hi Guys, same here but I have the Logs and the Project,

he is seemingly killing himself early on in the game, trying to get the manufacturer from zigbee?!, 

but no earlier meaningful logmessages?!

hopefully this meight help.

PS: tried it with IDF Version 5.3 too, with the same result.

[16:56:51]ELF file SHA256: a5191f47
[16:56:51]
[16:56:51]Rebooting...
[16:56:51]ESP-ROM:esp32c6-20220919
[16:56:51]Build:Sep 19 2022
[16:56:51]rst:0xc (SW_CPU),boot:0x6e (SPI_FAST_FLASH_BOOT)
[16:56:51]Saved PC:0x4001975a
[16:56:51]SPIWP:0xee
[16:56:51]mode:DIO, clock div:2
[16:56:51]load:0x4086c410,len:0xd8c
[16:56:51]load:0x4086e610,len:0x2e1c
[16:56:51]load:0x40875728,len:0x17d0
[16:56:51]entry 0x4086c410
[16:56:51]I (22) boot: ESP-IDF 5.1.5 2nd stage bootloader
[16:56:51]I (22) boot: compile time Dec 29 2024 16:53:30
[16:56:51]I (24) boot: chip revision: v0.0
[16:56:51]I (25) boot.esp32c6: SPI Speed      : 80MHz
[16:56:51]I (29) boot.esp32c6: SPI Mode       : DIO
[16:56:51]I (34) boot.esp32c6: SPI Flash Size : 4MB
[16:56:51]I (39) boot: Enabling RNG early entropy source...
[16:56:51]I (45) boot: Partition Table:
[16:56:51]I (48) boot: ## Label            Usage          Type ST Offset   Length
[16:56:51]I (55) boot:  0 otadata          OTA data         01 00 00009000 00002000
[16:56:51]I (63) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[16:56:51]I (70) boot:  2 app0             OTA app          00 10 00010000 001b0000
[16:56:51]I (78) boot:  3 app1             OTA app          00 11 001c0000 001b0000
[16:56:51]I (85) boot:  4 nvs              WiFi data        01 02 00370000 0006d000
[16:56:51]I (92) boot:  5 zb_storage       Unknown data     01 81 003dd000 00004000
[16:56:51]I (100) boot:  6 zb_fct           Unknown data     01 81 003e1000 00000400
[16:56:51]I (108) boot: End of partition table
[16:56:51]I (112) esp_image: segment 0: paddr=00010020 vaddr=42080020 size=114a8h ( 70824) map
[16:56:51]I (148) esp_image: segment 1: paddr=000214d0 vaddr=40800000 size=0ca18h ( 51736) load
[16:56:51]I (173) esp_image: segment 2: paddr=0002def0 vaddr=4080ca20 size=01fe0h (  8160) load
[16:56:51]I (179) esp_image: segment 3: paddr=0002fed8 vaddr=00000000 size=00140h (   320)
[16:56:51]I (181) esp_image: segment 4: paddr=00030020 vaddr=42000020 size=72288h (467592) map
[16:56:51]I (374) boot: Loaded app from partition at offset 0x10000
[16:56:51]I (375) boot: Disabling RNG early entropy source...
[16:56:51]I (386) cpu_start: Unicore app
[16:56:51]I (387) cpu_start: Pro cpu up.
[16:56:51]W (396) clk: esp_perip_clk_init() has not been implemented yet
[16:56:51]I (403) cpu_start: Pro cpu start user code
[16:56:51]I (403) cpu_start: cpu freq: 160000000 Hz
[16:56:51]I (404) cpu_start: Application information:
[16:56:51]I (406) cpu_start: Project name:     maingaragetest
[16:56:51]I (412) cpu_start: App version:      2024.12.2
[16:56:51]I (417) cpu_start: Compile time:     Dec 29 2024 16:53:18
[16:56:51]I (423) cpu_start: ELF file SHA256:  a5191f47...
[16:56:51]I (428) cpu_start: ESP-IDF:          5.1.5
[16:56:51]I (433) cpu_start: Min chip rev:     v0.0
[16:56:51]I (438) cpu_start: Max chip rev:     v0.99
[16:56:51]I (442) cpu_start: Chip rev:         v0.0
[16:56:51]I (447) heap_init: Initializing. RAM available for dynamic allocation:
[16:56:51]I (454) heap_init: At 408132B0 len 00069360 (420 KiB): D/IRAM
[16:56:51]I (461) heap_init: At 4087C610 len 00002F54 (11 KiB): STACK/DIRAM
[16:56:51]I (468) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
[16:56:51]I (475) spi_flash: detected chip: generic
[16:56:51]I (479) spi_flash: flash io: dio
[16:56:51]I (483) sleep: Configure to isolate all GPIO pins in sleep state
[16:56:51]I (489) sleep: Enable automatic switching of GPIO sleep configuration
[16:56:51]I (496) coexist: coex firmware version: 27d8387
[... 109 lines cut ...]
  - platform: gpio
    pin:
      number: 5
      mode:
        input: true
        pullup: true
      inverted: true
    id: door_closed_sensor
    name: "Gartentor geschlossen Sensor"
  - platform: gpio
    pin:
      number: 9
      mode:
        input: true
        pullup: true
      inverted: true
    id: button_1
    on_press:
      then:
        - zigbee.report: zb
    on_click:
      min_length: 5s
      max_length: 20s
      then:
        - zigbee.reset: zb

luar123, 2024-12-29:
I don't think that is the same issue. Anyway, in your case you probably activated reporting for an attribute that is not reportable. Try to set `reporting: false`
Please read ZCL.
If I get your usecase right, a binary sensor, you should use binary input not output (you can use simple_sensor instead of on_off_output) and you need to set the correct attribute ID, I don't think it is 0.

closed 2025-01-24 by 10der