luar123/zigbee_esphome #21: Setting attribute failed: ERROR

DustinThal, 2025-01-17:
Hi, 

first of all, thanks for your good work. with your repository I got Zigbee on Xiao esp32c6 working. (sometimes in combination with wifi)
But I have a litte question, cause I am lost.
attached is my yaml part, the log and a picture form z2m

why is my pm25 value 0? shouldn't it be 4 or 5 ?.

and is there an other attribute I can use instead of 4 times pm25 ?

This is part of my yaml:
```
sensor:
  - platform: sen5x
    id: SEN55
    pm_1_0:
      name: " PM <1µm Weight concentration"
      id: air_pm_1_0
      #accuracy_decimals: 1
    pm_2_5:
      name: " PM <2.5µm Weight concentration"
      id: air_pm_2_5
      #accuracy_decimals: 1
    pm_4_0:
      name: " PM <4µm Weight concentration"
      id: air_pm_4_0
      #accuracy_decimals: 1
    pm_10_0:
      name: " PM <10µm Weight concentration"
      id: air_pm_10_0
      #accuracy_decimals: 1
    temperature:
      name: "Temperature"
      id: air_temperature
      #accuracy_decimals: 1
    humidity:
      name: "Humidity"
      id: relative_humidity
      #accuracy_decimals: 1
    voc:
      name: "VOC"
      algorithm_tuning:
        index_offset: 100
        learning_time_offset_hours: 12
        learning_time_gain_hours: 12
        gating_max_duration_minutes: 180
        std_initial: 50
        gain_factor: 230
    nox:
      name: "NOX"
      algorithm_tuning:
        index_offset: 1
        learning_time_offset_hours: 12
        learning_time_gain_hours: 12
        gating_max_duration_minutes: 720
        std_initial: 50
        gain_factor: 230 
    temperature_compensation:
      offset: 0
      normalized_offset_slope: 0
      time_constant: 0
    acceleration_mode: low
    store_baseline: true
    address: 0x69
    update_interval: 10s

zigbee:
  id: "zb"
  endpoints:
    - device_type: TEMPERATURE_SENSOR
[... 330 lines cut ...]
[23:11:42][VV][i2c.idf:173]: 0x69 RX 002C7A002E18002E18002E1812CC0C121D3903F24C000A5A
[23:11:42][V][sensor:043]: ' PM <1µm Weight concentration': Received new state 4.400000
[23:11:42][D][sensor:093]: ' PM <1µm Weight concentration': Sending state 4.40000 µg/m³ with 2 decimals of accuracy
[23:11:42][V][sensor:043]: ' PM <2.5µm Weight concentration': Received new state 4.600000
[23:11:42][D][sensor:093]: ' PM <2.5µm Weight concentration': Sending state 4.60000 µg/m³ with 2 decimals of accuracy
[23:11:42][V][sensor:043]: ' PM <4µm Weight concentration': Received new state 4.600000
[23:11:42][D][sensor:093]: ' PM <4µm Weight concentration': Sending state 4.60000 µg/m³ with 2 decimals of accuracy
[23:11:42][V][sensor:043]: ' PM <10µm Weight concentration': Received new state 4.600000
[23:11:42][D][sensor:093]: ' PM <10µm Weight concentration': Sending state 4.60000 µg/m³ with 2 decimals of accuracy
[23:11:42][V][sensor:043]: 'Temperature': Received new state 23.184999
[23:11:42][D][sensor:093]: 'Temperature': Sending state 23.18500 °C with 2 decimals of accuracy
[23:11:42][V][sensor:043]: 'Humidity': Received new state 48.119999
[23:11:42][D][sensor:093]: 'Humidity': Sending state 48.12000 % with 2 decimals of accuracy
[23:11:42][V][sensor:043]: 'VOC': Received new state 101.000000
[23:11:42][D][sensor:093]: 'VOC': Sending state 101.00000  with 0 decimals of accuracy
[23:11:42][V][sensor:043]: 'NOX': Received new state 1.000000
[23:11:42][D][sensor:093]: 'NOX': Sending state 1.00000  with 0 decimals of accuracy
[23:11:42][D][zigbee:018]: Attribute set!
[23:11:42][D][zigbee:018]: Attribute set!
[23:11:42][D][zigbee:018]: Attribute set!
[23:11:42][D][zigbee:018]: Attribute set!
```

Picture from Z2M
[image]

luar123, 2025-01-18:
I had the same issue and usually the answer is in the ZCL definition:
pm25 meas cluster attribute 0 is the measured value in ppm between 0 and 1 and type is single. However, all sensors that I could find report directly `µg/m³` and this is also what z2m expects. So what you need to do is, setup attribute 0 without scaling and correct type and additionally you need to set attribut 2 (max measured value) to the maximum value your sensor is able to measure (or higher). For me this looks like:
```
- id: PM2_5_MEASUREMENT
          attributes:
            - attribute_id: 0x0
              type: single
              report: true
              value: 0
              device: air_pm_2_5
            - attribute_id: 0x2
              type: single
              value: 999
```

DustinThal, 2025-01-18:
big thanks, its working.
i will have a closer look into ZCL definitions, maybe I find something for Custom  "- id:" so I don't have to change the title of the measurement in HA. Many thanks again, hung on that several hours.

[image]

luar123, 2025-01-18:
Custom clusters are supported, just use an id >= 0xFC00:
```
- id: 0xFC2c
          attributes:
            - attribute_id: 0x0
              type: single
              report: true
              access: read_only
              value: 0
              device: pm10
            - attribute_id: 0x2
              type: single
              access: read_only
              value: 999
```
(attribute 2 won't be needed I guess)
And then you have to add a custom converter to z2m (I have not done that yet): https://www.zigbee2mqtt.io/advanced/support-new-devices

DustinThal, 2025-01-19:
thanks for your help, got some errors cause PM25 is single, but z2m things it must be uint16 or so.
I put pm25 also into custom and know no errors anymore

if someone is interested:

```
esphome:
  name: esphome-sen55
  friendly_name: esphome-sen55
  platformio_options:
    platform: https://github.com/lucaskatayama/platform-espressif32.git#feat/seeed_xiao_esp32c6.json
    build_flags: "-DI2C_NUM_1=I2C_NUM_0"
    #board_build.flash_mode: dio
    #board_build.maximum_size: 1310720

esp32:
  board: seeed_xiao_esp32c6
  variant: ESP32C6
  partitions: partitions_zb.csv
  framework:
    type: esp-idf
    version: 5.3.0
    #platform_version: 6.9.0
    sdkconfig_options:
      #CONFIG_ESPTOOLPY_FLASHSIZE_4MB: y
      #CONFIG_ESPTOOLPY_FLASHSIZE_8MB: y
      #CONFIG_BT_BLE_50_FEATURES_SUPPORTED: y
      #CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y
      CONFIG_ESP_TASK_WDT: y
      CONFIG_ESP_TASK_WDT_TIMEOUT_S: "10"
      CONFIG_OPENTHREAD_ENABLED: n
      CONFIG_ENABLE_WIFI_STATION: y
      CONFIG_USE_MINIMAL_MDNS: y

external_components:
  - source: components
    components: [zigbee]

# Enable logging
logger:
  hardware_uart: USB_SERIAL_JTAG
#  level: VERY_VERBOSE

#wifi:
#  ssid: !secret wifi_ssid
#  password: !secret wifi_password

#ota:
#  platform: esphome

#web_server:
#  port: 80

i2c:
  sda: GPIO22
  scl: GPIO23
  scan: false
  id: bus_0x69

sensor:
  - platform: sen5x
    id: SEN55
    pm_1_0:
      name: " PM <1µm Weight concentration"
      id: air_pm_1_0
    pm_2_5:
      name: " PM <2.5µm Weight concentration"
      id: air_pm_2_5
    pm_4_0:
      name: " PM <4µm Weight concentration"
      id: air_pm_4_0
    pm_10_0:
      name: " PM <10µm Weight concentration"
      id: air_pm_10_0
    temperature:
      name: "Temperature"
      id: air_temperature
    humidity:
      name: "Humidity"
      id: relative_humidity
    voc:
      name: "VOC"
      id: air_voc
      algorithm_tuning:
        index_offset: 100
        learning_time_offset_hours: 12
        learning_time_gain_hours: 12
        gating_max_duration_minutes: 180
        std_initial: 50
        gain_factor: 230
    nox:
      name: "NOX"
      id: air_nox
      algorithm_tuning:
        index_offset: 1
        learning_time_offset_hours: 12
        learning_time_gain_hours: 12
        gating_max_duration_minutes: 720
        std_initial: 50
        gain_factor: 230 
    temperature_compensation:
      offset: 0
      normalized_offset_slope: 0
      time_constant: 0
    acceleration_mode: low
    store_baseline: true
    address: 0x69
    update_interval: 10s

zigbee:
  id: "zb"
  endpoints:
    - device_type: TEMPERATURE_SENSOR
      num: 1
      clusters:
        - id: TEMP_MEASUREMENT
          attributes:
            - attribute_id: 0x0
              type: S16
              report: true
              value: 100
              device: air_temperature
              scale: 100
        - id: REL_HUMIDITY_MEASUREMENT
          attributes:
            - attribute_id: 0x0
              type: S16
              report: true
              scale: 100
              device: relative_humidity

    - device_type: CUSTOM_ATTR
      num: 2
      clusters:
        - id: 0xFC2C
          attributes:
            - attribute_id: 0x0
              type: U16
              report: true
              access: read_only
              value: 0
              scale: 100
              device: air_pm_1_0
        - id: 0xFC2D 
          attributes:
            - attribute_id: 0x0
              type: U16
              report: true
              access: read_only
              value: 0
              scale: 100
              device: air_pm_2_5
        - id: 0xFC2E
          attributes:
            - attribute_id: 0x0
              type: U16
              report: true
              access: read_only
              value: 0
              scale: 100
              device: air_pm_4_0
        - id: 0xFC2F
          attributes:
            - attribute_id: 0x0
              type: U16
              report: true
              access: read_only
              value: 0
              scale: 100
              device: air_pm_10_0
        - id: 0xFC30
          attributes:
            - attribute_id: 0x0
              type: U16
              report: true
              access: read_only
              value: 0
              scale: 100
              device: air_voc
        - id: 0xFC31
          attributes:
            - attribute_id: 0x0
              type: U16
              report: true
              access: read_only
              value: 0
              scale: 100
              device: air_nox

  on_join:
    then:
      - logger.log: "Joined network"
```

closed 2025-01-19 by DustinThal