luar123/zigbee_esphome #27: Various startup errors when setting up electrical measurement and metering cluster

papylhomme, 2025-01-27:
Hello,

I'm trying to set up reporting from a PZEM-004T sensor using zigbee. After digging into the ZCL specs I came up with this config, and it's more or less working.

## Working ✅ 
voltage, power and current are exposed in Z2M, read command works, reporting works too 

[image]

[image]

## Almost Working 👷 

For frequency, read command and reporting works, but somehow the sensor is not directly exposed in Z2M 🤷 If you get any idea about this, otherwise I can live without it.

Expose is missing frequency...
[image]

...but reading frequency is working:
[image]

## Not really working 🚧 
For Current Summation Delivered it's not working so easily

Sensor is exposed with no associated value...
[image]

...but I can read the value using a command ✅ 
[image]

Z2M complains about missing attributes in the metering cluster when configuring the device, and also querying them directly (divisor/multiplier), although they are present in the config (see below)

[image]

Unable to configure reporting with the error "UNREPORTABLE_ATTRIBUTE". This is clearly not quite true, I checked other zigbee appliance and they have reporting enabled for this attribute ([Adeo SIN-4-FP-21_EQU](https://www.zigbee2mqtt.io/devices/SIN-4-FP-21_EQU.html), [Nous A1Z](https://www.zigbee2mqtt.io/devices/A1Z.html))
[image]

## TODO

Better control with formatting, right now I'm missing some decimals.

## Startup errors

- I can see there is an ESP_FAIL error when adding seMetering divisor and multiplier attribute (0x301 and 0x302) => Explain the error in Z2M
- Many "ESP_ZIGBEE_ATTRIBUTE: The requested update attribute is not found"
- For each cluster "ESP_ZIGBEE_ATTRIBUTE: The requested update attribute ID:0x508 is not found"

Here is the output when starting the device : 

```
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyACM0 with baud rate 115200
[22:27:02]Vj
[22:27:02]load:0x408460e8,len:0x1778
[22:27:02]entry 0x4083cfd0
[22:27:02]I (23) boot: ESP-IDF 5.1.5 2nd stage bootloader
[22:27:02]I (24) boot: compile time Jan 24 2025 18:53:23
[22:27:02]I (25) boot: chip revision: v0.1
[22:27:02]I (26) boot.esp32h2: SPI Speed      : 64MHz
[22:27:02]I (31) boot.esp32h2: SPI Mode       : DIO
[22:27:02]I (36) boot.esp32h2: SPI Flash Size : 4MB
[22:27:02]I (41) boot: Enabling RNG early entropy source...
[22:27:02]I (46) boot: Partition Table:
[22:27:02]I (50) boot: ## Label            Usage          Type ST Offset   Length
[22:27:02]I (57) boot:  0 otadata          OTA data         01 00 00009000 00002000
[22:27:02]I (64) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[22:27:02]I (72) boot:  2 app0             OTA app          00 10 00010000 001b0000
[22:27:02]I (79) boot:  3 app1             OTA app          00 11 001c0000 001b0000
[22:27:02]I (87) boot:  4 nvs              WiFi data        01 02 00370000 0006d000
[22:27:02]I (94) boot:  5 zb_storage       Unknown data     01 81 003dd000 00004000
[... 267 lines cut ...]
            - attribute_id: 0x0200 # Status
              type: 8BITMAP
              report: false
              value: 0 # No error to report
            - attribute_id: 0x0300 # UnitofMeasure
              type: 8BIT_ENUM
              report: false
              value: 0x00 # kWh (Kilowatt Hours) & kW (Kilowatts) in pure binary format
            - attribute_id: 0x0301 # Multiplier
              type: U24
              report: false
              value: 1
            - attribute_id: 0x0302 # Divisor
              type: U24
              report: false
              value: 1
            - attribute_id: 0x0303 # SummationFormatting
              type: 8BITMAP
              report: false
              value: 0b11111111
            - attribute_id: 0x0306 # MeteringDeviceType
              type: 8BITMAP
              report: false
              value: 0 # Electric Metering
```

papylhomme, 2025-01-29:
> Some quick comments:
> 
> * missing exposes: maybe not auto detected and you have to create a custom converter

I'll look into that. For clarification, you mean something like [this](https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/externalConvertersExample/sensor.js) ? It's weird the others are exposed "by default" and not this one, I'll ask on their channels if there is an explanation for this.

> * `ESP_ZIGBEE_ATTRIBUTE: The requested update attribute` can be ignored

Ok

> * metering cluster attribute 0 does not support reporting according to ZCL. You could try to overwrite by adding `access: read_only` (not sure if the sdk allows that).

When you say it does not support reporting according to the specs, you mean RP does not appear in the access column of this attribute ? If so, none of the metering cluster attributes does, and still reporting is working well for the others.

Anyhow, I think the `UNREPORTABLE_ATTRIBUTE` error in Z2M is caused by the missing reporting info in the H2. I tried to add some debug message for the error `Could not configure reporting for attribute 0x0000 in cluster 0x0702 in endpoint 1`, and the result code from the `esp_zb_zcl_update_reporting_info` call is `ESP_ERR_NO_MEM (0x0101)`.

So I tried removing the other reports to see if it would free up some memory but the error is the same. I'm thinking it may be related to the size of this attribute (uint48) exceeding some buffer allocated for the reporting info, but I could not find the implementation of `esp_zb_zcl_update_reporting_info` to check that. Do you know where I can ask for clarification on this one ?

> * `ESP_Fail` errors are real. You need to add `access: read_only`. Not sure why the default approach is not implemented for metering cluster. I should add a test to the config validation.

Thanks it did the trick : I can query divisor and multiplier from Z2M, and there is no more error when configuring the device.

luar123, 2025-01-30:
> I'll look into that. For clarification, you mean something like [this](https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/externalConvertersExample/sensor.js) ? It's weird the others are exposed "by default" and not this one, I'll ask on their channels if there is an explanation for this.

Yes, that is what I meant. But actually I am not so sure anymore, seems to be supported: https://github.com/Koenkk/zigbee-herdsman-converters/blob/09cfce5153d3931c7cc009aa3ae3107d9b394f5c/src/lib/modernExtend.ts#L1665
Is there anything in z2m logs? Could also be a configuration issue. Some needed attributes missing or so.
Try to set `acFreqMin=0` and `acFreqMax=1000`.

> When you say it does not support reporting according to the specs, you mean RP does not appear in the access column of this attribute ? If so, none of the metering cluster attributes does, and still reporting is working well for the others.

Indeed, I though non of the metering attributes support reporting. In general the metering cluster seems very different from all the other sensor clusters. It also supports server command, snapshots, mirrors, etc.

> Anyhow, I think the `UNREPORTABLE_ATTRIBUTE` error in Z2M is caused by the missing reporting info in the H2. I tried to add some debug message for the error `Could not configure reporting for [...]

You can ask at the sdk issue tracker, but I guess they will tell you that reporting is not supported for that attribute, like here: https://github.com/espressif/esp-zigbee-sdk/issues/463

papylhomme, 2025-02-15:
> Yes, that is what I meant. But actually I am not so sure anymore, seems to be supported: [...]

Tried adding `acFreqMin` and `acFreqMax` but dit not change anything, neither did playing with the *basic information* flag.
Eventually  tried the custom converter road and it works when forcing the frequency in the args. When looking to the `ElectricityMeterArgs`, it seems power/voltage/current are enabled by default (only possible to disable them), but acFrequency needs to be forced.

```js
const m = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['esp-h2-power-monitoring-1'],
    model: 'esp-h2-power-monitoring-1',
    vendor: 'esphome',
    description: 'Zigbee power metering',
    extend: [m.electricityMeter({acFrequency: true})],
    meta: {},
};

module.exports = definition;
```

> Indeed, I though non of the metering attributes support reporting. In general the metering cluster seems very different from all the other sensor clusters. It also supports server command, snapshots, mirrors, etc.
> You can ask at the sdk issue tracker, but I guess they will tell you that reporting is not supported for that attribute, like here: [espressif/esp-zigbee-sdk#463](https://github.com/espressif/esp-zigbee-sdk/issues/463)

I'll try that, but I need to write a minimal example first, might take some time as I use esphome to skip this part altogether .. 😄