luar123/zigbee_esphome #28: Can't configure power source on initial network enrollment

Sharp398, 2025-01-30:
Hi there,

I'm trying to setup an ESP32-C6 as a battery-based device with deep-sleep to initialize once every hour to take some measurements, including battery state, and report it to HA via ZHA.

I've gotten the device to boot and successfully enroll in the ZHA network and report things like temperature and humidity from an AHT20. However, in Home Assistant the device does not appear to display any battery entity. I suspect it's because the ZHA integration reports the Power Source is Mains, so HA does not assume there's a battery entity.

[image]

I've tried to configure the `power_supply: 0x03` under the `zigbee` configuration in the esphome yaml file, as the ZCL reports that 0x03 is the enum for battery. I've even tried to manually configure the `BASIC` cluster with the Power Source attribute and the `POWER_CONFIG` as follows:
```
        - id: POWER_CONFIG                                                                                                 
          attributes:                                                                                                      
            - attribute_id: 0x0020 # Battery Voltage                                                                       
              type: U8                                                                                                     
              report: true                                                                                                 
              id: v_batt                                                                                                   
              value: 255                                                                                                   
              access: read_only                                                                                            
            - attribute_id: 0x0021 # Battery Percentage                                                                    
              type: U8                                                                                                     
              report: true                                                                                                 
              id: batt_percent                                                                                             
              value: 255                                                                                                   
              access: read_only
            - attribute_id: 0x0031 # Battery Size
              type: 8BIT_ENUM
              value: 2 # Other 
              access: read_write
            - attribute_id: 0x0033 # Battery Quantity
              type: U8
              value: 1
              access: read_write
            - attribute_id: 0xfffd # Cluster Revision
              type: U16
              value: 6
              access: read_only
        - id: BASIC
          attributes:
            - attribute_id: 0x0007 # Power Source
              type: 8BIT_ENUM
              value: 3 # Battery
              access: read_only
```
Even if I remove the `BASIC` cluster configuration and leave just the `power_supply: 0x03`, the "Manage zigbee device" dialog lets me query the `power_source` attribute of the `BASIC` cluster and yields `PowerSource.Battery`, and if I remove the `power_supply: 0x03`, it'll now yield `PowerSource.Unknown`.

Help me understand what needs to happen so that ZHA will report this device as a battery device and successfully display the battery entity data for it.

Sharp398, 2025-01-30:
@simmsb Thanks! I finally have battery percentage reported in ZHA!

I initially tried to just copy your `sdkconfig_options` but that wasn't enough. It only finally worked when I used your github repo as the source for the zigbee component, instead of this one. 

I see one commit you have to `zb_set_ed_node_descriptor` which does have a `bool power_src` variable in the signature. Can you describe what that change is doing? Given it solved my problem, seems like it'd be valuable to merge it back into this repo so others may benefit.

Sharp398, 2025-01-30:
Hmm, I'm getting errors like this now:
```
[E][zigbee:119][Zigbee_main]: FIRST_START.  Device started up in non factory-reset mode with an error -1 (ESP_FAIL)        
[W][zigbee:122][Zigbee_main]: Failed to initialize Zigbee stack (status: ESP_FAIL)
```
Specifically when I try to take your change and merge it into this repo, but also even if I'm still using your fork it's started to do this seemingly inexplicably. Using this repo directly still works fine, but doesn't give me back battery voltage. Not sure what's changed, I've even tried running `esphome clean` first.

Seeing logs like this further up as well:
```
[D][esp-idf:000]: E (527) ESP_ZIGBEE_ATTRIBUTE: The requested update attribute ID:0x21 is not found                        
                                                                                                                           
[D][esp-idf:000]: E (532) ESP_ZIGBEE_ATTRIBUTE: The requested update attribute ID:0x31 is not found
...
[D][esp-idf:000][Zigbee_main]: I (667) phy_init: phy_version 310,dde1ba9,Jun  4 2024,16:38:11                              
                                                                                                                           
[D][esp-idf:000][Zigbee_main]: I (719) phy: libbtbb version: 04952fd, Jun  4 2024, 16:38:26                                
                                                                                                                           
[D][esp-idf:000][Zigbee_main]: E (720) ESP_ZIGBEE_CLUSTER: The requested update cluster ID (0x   1) is not found           
                                                                                                                           
[D][esp-idf:000][Zigbee_main]: E (720) zigbee_helper: Ignore previous cluster not found error                              
                                                                                                                           
[D][esp-idf:000][Zigbee_main]: E (729) ESP_ZIGBEE_CLUSTER: The requested update cluster ID (0x 405) is not found           
                                                                                                                           
[D][esp-idf:000][Zigbee_main]: E (739) zigbee_helper: Ignore previous cluster not found error                              
                                                                                                                           
[I][zigbee:356][Zigbee_main]: set reporting for cluster: 1029                                                              
[I][zigbee:356][Zigbee_main]: set reporting for cluster: 1026                                                              
[I][zigbee:356][Zigbee_main]: set reporting for cluster: 1                                                                 
[C][gpio.output:010]: GPIO Binary Output:                                                                                  
[C][gpio.output:011]:   Pin: GPIO5                                                                                         
[I][zigbee:182][Zigbee_main]: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL                                        
[I][zigbee:094][Zigbee_main]: SKIP_STARTUP. Device started up in non factory-reset mode                                    
[I][zigbee:096][Zigbee_main]: Initialize Zigbee stack
...
[C][zigbee:410]: ZigBee:                                                                                                   
[C][zigbee:412]: Endpoint: 1, 770                   
```

luar123, 2025-01-30:
I didn't look into battery powered devices yet, but I think there is a bug that needs a fix with the `zb_set_ed_node_descriptor` function: https://github.com/espressif/arduino-esp32/issues/10612
There is no real documentation for that function.

So setting `power_supply: 0x03` (or manipulating the BASIC cluster directly) and enabling battery reporting needs some further work. You could try: `zb_set_ed_node_descriptor(0, true, 1)` at the same position as @simmsb. That is what they are using in Arduino.

I would try to remove all `access: read_only` parameters, those should be used only with custom clusters/attributes and a few special clusters like metering.
Also always re-interview and re-join your device when changing the zigbee configuration.
Oh, and sometimes it helps to run `esptool.py erase_flash`

simmsb, 2025-01-31:
Yeah, I think I encountered similar issues (at least when turning off rx_on_when_idle) and a combination of removing (/force removing) the device from the zigbee network and factory resetting it would allow it to connect again. I think the issue is rx_on_when_idle while the device is bound to the network.

Sharp398, 2025-02-01:
Indeed, running `erase_flash` helps to reset it and get it working again. I'll try to keep working with this now, I've gotten the battery measurements I need into HA so I'm on the right track now.

closed 2025-08-16 by luar123