luar123/zigbee_esphome #33: Reporting of either ENUM8 type or RW access attributes does not work

baal86, 2025-02-22:
Experimenting with fan control clusters 0x0202 and the fan mode attribute 0x0000 which is ENUM8 type and RW access. Could not find example code for either ENUM8 or RW access.

The issue is that if I setAttr the fan mode attribute and then force report it does not update on the far end although the logs claim tthat the attribute was successfully set. I suspect it is either the ENUM8 data type or the RW access that causes this issue. Receiving the attribute works fine. The LOG entry is successfully triggered every time I change state from HA.

I have inserted my code below.

**Expected Behaviour:**
Every second both the pressure sensor (as "placebo") value and fan_mode value get reported. Receiving fan_mode works.

** Actual Behaviour:**
The pressure sensor gets reported correctly while the fan mode never gets reported. Receiving fan_mode works.

```
esphome:
  name: zb-fan-c6
  friendly_name: zb-fan-c6

esp32:
  board: esp32-c6-devkitm-1
  partitions: partitions_zb.csv
  framework:
    type: esp-idf
    version: latest

external_components:
  - source:
      type: git
      url: https://github.com/luar123/zigbee_esphome
      ref: "master"
    refresh: 0s
    components: [zigbee]

logger:
  hardware_uart: USB_SERIAL_JTAG
  level: DEBUG

sensor:
  - platform: internal_temperature
    name: "Internal Temperature"
    id: "temp"
    filters:
      - delta: 0.1
  - platform: template
    name: "Template Sensor"
    id: "press"
    lambda: "return int(rand() % 100 + 100);"
    update_interval: 1s
    on_value: 
      then:
        - zigbee.setAttr:
            id: pressure_attr
            value: !lambda "return int16_t(x);"
        - zigbee.setAttr:
            id: fan_mode_attr
            value: !lambda "return uint8_t(1);"
        - zigbee.report

zigbee:
  id: "zb"
  endpoints:
    - device_type: TEMPERATURE_SENSOR
      num: 1
      clusters:
        - id: TEMP_MEASUREMENT
          attributes:
            - attribute_id: 0x0
              type: S16
              report: true
              value: 100
              device: temp
              scale: 100
        - id: PRESSURE_MEASUREMENT
          attributes:
            - attribute_id: 0
              id: pressure_attr
              type: S16
              report: true
              value: 100
        - id: FAN_CONTROL
          attributes:
            - attribute_id: 0x0000
              id: fan_mode_attr
              type: 8BIT_ENUM
              report: true
              value: 1
              access: "READ_WRITE"
              on_value:
                then:
                  logger.log: 
                    format: "Received state %d" 
                    args: [!lambda "x"]

binary_sensor:
  - platform: gpio
    pin:
      number: 9
      mode:
        input: true
        pullup: true
      inverted: true
    id: button_1
    on_click:
      min_length: 3s
      max_length: 20s
      then:
        - zigbee.reset
 ```

baal86, 2025-02-23:
Found it! There's indeed an error message.

```
[15:51:41][E][zigbee:481]: Could not configure reporting for attribute 0x0000 in cluster 0x0202 in endpoint 1
```

I had naively assumed that READ/WRITE would include reporting as well. Do you know how commercial products handle status readback? I am reasonably certain that if I change colour of my IKEA lights HA does not need to keep the state of that attribute but maybe I'm wrong there..

closed 2025-02-23 by baal86

luar123, 2025-02-23:
You need to check [ZCL](https://csa-iot.org/wp-content/uploads/2022/01/07-5123-08-Zigbee-Cluster-Library-1.pdf) for each cluster/attribute. Most of light/sensor attributes support reporting.
If reporting is not supported the coordinator need to read the attribute. Either regular polling or through some status attributes that are reported (I didn't check the full HVAC section to see what is available for HVAC/fan clusters).

baal86, 2025-02-23:
Yeah, had looked into the ZCL specs but funnily enough only at attributes / clusters that don't really use the RP or RWP mode! So only thing I had seen so far were R, W, RW. But that means that fan cluster is fully working!