luar123/zigbee_esphome #38: Trouble getting multiple DIMMABLE_LIGHT endpoints to work properly

gwmullin, 2025-04-07:
I have a config for controlling 5 LED strips via ledc and esphome monochromatic, however it doesn't actually appear to work correctly.

esphome shows all 5 lights, and they work correctly when controlled via esphome (local web page), however in zigbee although I have 5 separate endpoints they all seem to control the first endpoint when any are activated, and homeassistant shows only a single light, which seems to control all 5 endpoints, but actually only controls the first light

Any clues on how to get this working?

```
external_components:
  - source: github://luar123/zigbee_esphome
    components: [zigbee]

esp32:
  board: esp32-c6-devkitc-1
  partitions: partitions_zb.csv
  framework:
    type: esp-idf

esphome:
  name: zigbeepwm
  friendly_name: Zigbee Pwmm
  on_boot:
    - priority: 700.0
      then:
        - script.execute: startup_routine

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "<REDACTED>"

ota:
  - platform: esphome
    password: "<REDACTED>"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  
# Optional, Enable Web server; start internal webserver so it can be used stand-alone
web_server:
  port: 80

# Ensure we are using the built-in antenna
script:
  - id: startup_routine
    mode: single
    then:
      - output.turn_off: gpio_3
      - delay: 0.1s
      - output.turn_on: gpio_14

button:
  - platform: template
    name: ZBRESET
    on_press:
      then:
        - zigbee.reset: zb
  - platform: factory_reset
    name: Restart with Factory Default Settings

output:
  - platform: gpio
    pin: GPIO3
    id: gpio_3
  - platform: gpio
    pin: GPIO14
    id: gpio_14
  - platform: ledc 
    pin: GPIO00
    id: gpio_00
    frequency: "9765Hz"
  - platform: ledc 
    pin: GPIO01
    id: gpio_01
    frequency: "9765Hz"
  - platform: ledc 
    pin: GPIO02
    id: gpio_02
    frequency: "9765Hz"
  - platform: ledc
    pin: GPIO21
    id: gpio_21
    frequency: "9765Hz"
  - platform: ledc
    pin: GPIO22
    id: gpio_22
    frequency: "9765Hz"

light:
  - platform: monochromatic
    name: "00"
    output: gpio_00
    id: "light_1"
  - platform: monochromatic
    name: "1"
    output: gpio_01
    id: "light_2"
  - platform: monochromatic
    name: "2"
    output: gpio_02
    id: "light_3"
  - platform: monochromatic
    name: "21"
    output: gpio_21
    id: "light_4"
  - platform: monochromatic
    name: "22"
    output: gpio_21
    id: "light_5"

    
zigbee:
  id: "zb"

  endpoints:
    - num: 1
      device_type: DIMMABLE_LIGHT
      clusters:
        - id: ON_OFF
          attributes:
            - attribute_id: 0
              type: bool
              on_value:
                then:
                  - light.control:
                      id: "light_1"
                      state: !lambda "return (bool)x;"
        - id: LEVEL_CONTROL
          attributes:
            - attribute_id: 0
              type: U8
              on_value:
                then:
                  - light.control:
                      id: "light_1"
                      brightness: !lambda "return ((float)x)/255;"
    - num: 2
      device_type: DIMMABLE_LIGHT
      clusters:
        - id: ON_OFF
          attributes:
            - attribute_id: 1
              type: bool
              on_value:
                then:
                  - light.control:
                      id: "light_2"
                      state: !lambda "return (bool)x;"
        - id: LEVEL_CONTROL
          attributes:
            - attribute_id: 1
              type: U8
              on_value:
                then:
                  - light.control:
                      id: "light_2"
                      brightness: !lambda "return ((float)x)/255;"
    - num: 3
      device_type: DIMMABLE_LIGHT
      clusters:
        - id: ON_OFF
          attributes:
            - attribute_id: 2
              type: bool
              on_value:
                then:
                  - light.control:
                      id: "light_3"
                      state: !lambda "return (bool)x;"
        - id: LEVEL_CONTROL
          attributes:
            - attribute_id: 2
              type: U8
              on_value:
                then:
                  - light.control:
                      id: "light_3"
                      brightness: !lambda "return ((float)x)/255;"
    - num: 4
      device_type: DIMMABLE_LIGHT
      clusters:
        - id: ON_OFF
          attributes:
            - attribute_id: 3
              type: bool
              on_value:
                then:
                  - light.control:
                      id: "light_4"
                      state: !lambda "return (bool)x;"
        - id: LEVEL_CONTROL
          attributes:
            - attribute_id: 3
              type: U8
              on_value:
                then:
                  - light.control:
                      id: "light_4"
                      brightness: !lambda "return ((float)x)/255;"
    - num: 5
      device_type: DIMMABLE_LIGHT
      clusters:
        - id: ON_OFF
          attributes:
            - attribute_id: 4
              type: bool
              on_value:
                then:
                  - light.control:
                      id: "light_5"
                      state: !lambda "return (bool)x;"
        - id: LEVEL_CONTROL
          attributes:
            - attribute_id: 4
              type: U8
              on_value:
                then:
                  - light.control:
                      id: "light_5"
                      brightness: !lambda "return ((float)x)/255;"
  on_join:
    then:
      - logger.log: "Joined network"
 ```
Zigbee2MQTT view:
[image]

Homeassistant view:
[image]

luar123, 2025-05-25:
There might be multiple issues and/or bugs but the most obvious is that your `attribute_id` always needs to be `0`. This is the zigbee attribute ID in the cluster. Please change and try again (trigger interview, remove device). If it is still not working we need to dig deeper.

luar123, 2025-07-07:
It is a bug in z2m: https://github.com/Koenkk/zigbee2mqtt/issues/27925
Until fixed you could use ZHA or create an external definition.