luar123/zigbee_esphome #48: No longer compiles on 2025.7.0b1

JochenFriedrich, 2025-07-10:
On 2025.7.0b1, the component lo longer compiles:

Reading CMake configuration...
-- Building ESP-IDF components for target esp32c6
Invalid field "dependencies:espressif__esp-zboss-lib": Invalid component name
-- Configuring incomplete, errors occurred!

CMake Error at /config/.esphome/platformio/packages/framework-espidf/tools/cmake/build.cmake:552 (message):
  ERROR: Manifest is not valid

Call Stack (most recent call first):
  /config/.esphome/platformio/packages/framework-espidf/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:3 (project)

luar123, 2025-07-10:
Will try to replicate. Are you using some modifications in the esphome/ESP-IDF section of the yaml?
And did you try to delete the .esphome/build/<name> folder?

JochenFriedrich, 2025-07-10:
Yes, I deleted .esphome/build and .esphome/platformio after upgrading to 7.0b1.

This is my esp-idf part of my yaml:

```
esp32:
  board: esp32-c6-devkitc-1
  flash_size: 4MB
  partitions: partitions_zb.csv
  framework:
    type: esp-idf
    #version: 5.4.2
    #platform_version: 54.03.21
    #platform_version: "https://github.com/pioarduino/platform-espressif32.git#develop"
    sdkconfig_options:
      CONFIG_ESPTOOLPY_FLASHSIZE_4MB: y
    #  CONFIG_LOG_DEFAULT_LEVEL_NONE: "y"
    #  CONFIG_ESP_CONSOLE_NONE: "y"
    #  CONFIG_RMT_RECV_FUNC_IN_IRAM: "y"
    #  CONFIG_RMT_ISR_IRAM_SAFE: "y"
  variant: esp32c6
  
```

closed 2025-07-13 by luar123